mavlink/mavros #1064: Waypoint Mission : WP: upload failed: coordinate frame is not supported

DesmondZhong, 2018-07-24:
##This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I try to use `mavros/mission/push` to send waypoint tables. In `Waypoint.msg`, I found the following definition for `frame`. 

```
# see enum MAV_FRAME
uint8 frame
uint8 FRAME_GLOBAL = 0
uint8 FRAME_LOCAL_NED = 1
uint8 FRAME_MISSION = 2
uint8 FRAME_GLOBAL_REL_ALT = 3
uint8 FRAME_LOCAL_ENU = 4
```

However, during my test in SITL, only `FRAME_GLOBAL` and `FRAME_GLOBAL_REL_ALT` works. If I use values 1, 2, 4. I'll get the error `WP: upload failed: coordinate frame is not supported`.

If this is the case, I can only use global frames to send Waypoint Missions. In other words, I need to set the longitude and latitude of points. However, if I want the drone to move to the north by 5 meters, converting "5 meters" to a latitude change is not convenient. If I can use local frames, the `x` and `y` units are already in meters. 

Could we use local frames to send missions? If we couldn't at this point, is there any way to get around it and send position in the unit of meter?

Thanks in advance!

### MAVROS version and platform

Mavros: 0.26.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: ?3.7.1?

### Node logs

```
... logging to /home/desmond/.ros/log/9703c3ba-8f86-11e8-85f1-bea5f405bcc1/roslaunch-Intel-Aero-RTF-12408.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Intel-Aero-RTF:37474/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
[... 467 lines cut ...]
```
OK. I got messages from 1:1.

---
Received 6287 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 26, 30, 31, 32, 33, 74, 76, 83, 85, 230, 105, 111, 241, 242, 245
Exception in thread Thread-2 (most likely raised during interpreter shutdown):Exception in thread Thread-4 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 754, in run
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/registration.py", line 275, in start
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/registration.py", line 307, in run
  File "/usr/lib/python2.7/threading.py", line 203, in release
<type 'exceptions.TypeError'>: 'NoneType' object is not callable

Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/timer.py", line 235, in run
  File "/home/desmond/catkin_ws/src/mavros/mavros/scripts/checkid", line 90, in timer_cb
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/core.py", line 493, in signal_shutdown
  File "/usr/lib/python2.7/threading.py", line 946, in join
<type 'exceptions.TypeError'>: 'NoneType' object is not callable

```

DesmondZhong, 2018-07-26:
Interesting. I can't find any documentation of whether px4 supports it or not. I am gonna try Ardupilot and see if it works. I will update here. Thanks!

DesmondZhong, 2018-07-26:
@TSC21 It seems that Ardupilot only supports global frames for waypoints too. Do you have any recommendation to control copters in the local frame using Ardupilot? Thanks!

TSC21, 2018-07-26:
Use Offboard mode in PX4. Use Guided in Ardupilot.

DesmondZhong, 2018-07-26:
Thanks! I'll explore a bit more!

atharvasaraf, 2020-01-21:
Hello @DesmondZhong ! Were you able to use the local frames to feed waypoints to the PX4 or the Ardupilot?