mavlink/mavros #1066: source installation   rosinstall_generator failed

floor-cheetah, 2018-07-26:
floor-cheetah@floorcheetah-Lenovo-Rescuer-14ISK:~/demo$ sudo rosinstall_generator --rosdistro kinetic mavlink | tee /tmp/mavros.rosinstall
[sudo] floor-cheetah 的密码： 
Traceback (most recent call last):
  File "/usr/bin/rosinstall_generator", line 141, in <module>
    rc = main()
  File "/usr/bin/rosinstall_generator", line 129, in main
    upstream_version_tag=args.upstream, upstream_source_version=args.upstream_development)
  File "/usr/lib/python2.7/dist-packages/rosinstall_generator/generator.py", line 256, in generate_rosinstall
    wet_package_names, unreleased_repo_names = _get_packages_for_repos(distro_name, repo_names, source=upstream_source_version)
  File "/usr/lib/python2.7/dist-packages/rosinstall_generator/generator.py", line 93, in _get_packages_for_repos
    wet_distro = get_wet_distro(distro_name)
  File "/usr/lib/python2.7/dist-packages/rosinstall_generator/generator.py", line 222, in get_wet_distro
    _wet_distro = _get_wet_distro(distro_name)
  File "/usr/lib/python2.7/dist-packages/rosinstall_generator/distro.py", line 45, in get_distro
    index = get_index(get_index_url())
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 101, in get_index
    yaml_str = load_url(url)
  File "/usr/lib/python2.7/dist-packages/rosdistro/loader.py", line 59, in load_url
    raise URLError(str(e) + ' (%s)' % url)
urllib2.URLError: <urlopen error <urlopen error [Errno -3] Temporary failure in name resolution> (https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml)>

vooon, 2018-07-26:
You have a problem with internet names resolution.
And why sudo?

floor-cheetah, 2018-07-26:
how can I solve this problem

floor-cheetah, 2018-07-27:
Finally Ｉ　ｈave d solved tihs problem thank you everybody !!!!

closed 2018-07-27 by vooon

huangyipei-cell, 2020-04-27:
Can you talk to me how do you fix it? Thank you very much!

tiger-k, 2022-06-18:
`sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
curl -sSL 'http://keyserver.ubuntu.com/pks/lookup?op=get&search=0xC1CF6E31E6BADE8868B172B4F42ED6FBAB17C654' | sudo apt-key add -
sudo apt update`
and then 
`rosinstall_generator --rosdistro kinetic mavlink | tee /tmp/mavros.rosinstall`