mavlink/mavros #1068: Adding RTK support (RTCM messages forward)

AlexisTM, 2018-07-27:
### Issue details

To allow many copters to fly with the RTK, we need to forward the RTCM messages to all of them. We cannot have QGC running so we would like to send the RTCM messages from ROS.

https://github.com/AlexisTM/mavros/commit/6338bdef3359ff0dbfa7c2b3bbe6a889c7483dae

That specific message is handled differently in MAVLink v1 and v2 (zeroes removed). Is this implementation enough to have both working?

vooon, 2018-07-27:
No, you incorrectly implement fragmentation.

Message should looks following:
```
Header header
uint8[] data
```

Handler should:
1. Check data length, if it < message payload (180): goto 2 else 3
2. Copy data, set len, fill remaining with 0 (optional), set flags = (seq << 3) | 0;
3. Copy first 180 bytes to first message, set falgs = (seq << 3) | fragment << 1 | 1; repeat up to 4 times.

vooon, 2018-07-27:
Also you can use Header.seq as sequence counter, but appy `& 0x1f`.

vooon, 2018-07-27:
And data[] may vary up to 180*4 bytes. So add assert `m.data.size() <= 180*4`.

vooon, 2018-07-27:
Also i think better name plugin `navsat_rtk` or `gps_rtk`

AlexisTM, 2018-07-27:
I was thinking of a 3rd party node handling the fragmentation for fast prototyping.
But it makes indeed more sense is this is done on mavros side of the RTCM messages are up to 700 bytes

ArkadiuszNiemiec, 2018-07-30:
I have implemented a setup that configures an ublox RTK base, survey-in the position over set time and then sends RTCM3 messages over ROS2. On our vehicle, we have ROS2->ROS1 bridge and a ROS1 plugin that sends it to PX4 that handles the rest.
Please tell me if anything would be useful for you 😄

AlexisTM, 2018-07-30:
That would be awesome! Feel free to share it :)

ArkadiuszNiemiec, 2018-07-31:
@AlexisTM https://github.com/ArkadiuszNiemiec/rtcm_inject
...but I just realized that it is a very similar code that you have in your PR. The RTCM3 stream is published on ROS2 and bridged to ROS1 on both sides but it's on this repo. Do you need anything else?

AlexisTM, 2018-07-31:
It is indeed very similar; 

Could you share that ROS2 node that reads the RTCM3 messages? That will allow me to test it directly :)

AlexisTM, 2018-08-07:
#1069 has been merged

closed 2018-08-07 by AlexisTM