mavlink/mavros #1071: No mapping for distance sensor

RR2-IP2, 2018-08-09:
Hi,

I am trying to access data from a distance sensor (Leddar One Range Finder) measuring height using MAVROS.  I am able to view the data using QGroundControl (as my flight controller is PX4).

On the drone (with MAVROS installed), I tried the following commands:
`$ roscore`
`$ rosrun mavros mavros_node _fcu_url:=tcp://127.0.0.1:5760 _system_id:=2`

After this, I get the same error repeated:
`DS: no mapping for sensor id: 0, type: 1, orientation: 25`

I am able to check the ID, type and orientation in QGC which makes me sure that this is the Leddar One sensor.
I have tried changing the distance_sensor information in px4_config.yaml file but have had no luck (tried various orientations: PITCH_270, NONE and 25).

I have found that the error is generated from line 205 in this code: [distance_sensor.cpp](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/distance_sensor.cpp) but I can't figure out what is causing it.

I am only new to ROS so any help would be greatly appreciated.

TSC21, 2018-08-09:
Have you checked the configuration for it on the `px4_config.yaml`? If you don't config the respective parameters to access the data, you won't be able to get the data properly. So you need to add a propper map for it on the yaml file (you already have examples there). Also, any reason for changing the sys_id to 2? You also don't need to remove `rangfinder` from blacklist because that's APM compatible only.

RR2-IP2, 2018-08-14:
I have resolved my issue.  I created a launch file that specifically passed in "px4_config.yaml" as the config_yaml argument.   This initialized distance sensor correctly and published the data to a ROS topic as desired.

This thread may be closed.

closed 2018-08-17 by vooon