mavlink/mavros #1072: Mavros kinectic undefined symbol

MC-Bourguiba, 2018-08-10:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi, I have been using docker to setup ros and mavros on a container. It was working perfectly fine and then suddenly I got the error.
TLDR here is the error:
```
* /mavros/setpoi/opt/ros/kinetic/lib/mavros/mavros_node: symbol lookup error: /opt/ros/kinetic/lib/libmavros.so: undefined symbol: _ZN17rosconsole_bridge26RegisterOutputHandlerProxyD1Ev
```
The Dockerfile is the following:
```
FROM ros:kinetic-ros-base

# Install mavros and rosbridge packages
RUN apt-get update && apt-get install -y \
    ros-kinetic-mavros \
    ros-kinetic-mavros-extras \
    ros-kinetic-rosbridge-suite \
    ros-kinetic-ros-tutorials \
    ros-kinetic-common-tutorials \
    && rm -rf /var/lib/apt/lists/

COPY ./launch /launch

RUN wget https://raw.githubusercontent.com/mavlink/mavros/master/mavros/scripts/install_geographiclib_datasets.sh

RUN chmod +x install_geographiclib_datasets.sh

RUN ./install_geographiclib_datasets.sh
```

Docker-compose file:
```
`version: '2'
services:
  master:
    build: ./ros
    container_name: master
    command:
      - roscore

  bridge:
    build: ./ros
    container_name: bridge
    ports:
      - "9090:9090"
    depends_on:
      - master
      - mavros
    environment:
      - "ROS_HOSTNAME=bridge"
      - "ROS_MASTER_URI=http://master:11311"
    command: roslaunch rosbridge_server rosbridge_websocket.launch --wait

  mavros:
    build: ./ros
    container_name: mavros
    depends_on:
      - master
    ports:
      - "14551:14551/udp"
      - "14550:14550/udp"
    environment:
      - "ROS_HOSTNAME=mavros"
      - "ROS_MASTER_URI=http://master:11311"
    command: roslaunch mavros apm.launch fcu_url:=udp://mavros:14551@ --wait 
```
[... 143 lines cut ...]
mavros    | NODES
mavros    |   /
mavros    |     mavros (mavros/mavros_node)
mavros    | 
mavros    | ROS_MASTER_URI=http://master:11311
mavros    | 
mavros    | process[mavros-1]: started with pid [37]
mavros    | [mavros-1] killing on exit
mavros    | shutting down processing monitor...
mavros    | ... shutting down processing monitor complete
mavros    | done
mavros exited with code 0
```

### Diagnostics

```
Can't access it since I'm using docker and the container crashes.
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2018-08-12:
Please check what version of mavros installed. 0.18 pretty old, it should be 0.25 at least.

Note: datasets can be installed by `rosrun mavros install_geographiclib_datasets.sh`, it shipped with other mavros scripts.

mikeisted, 2018-08-14:
I have the same issue with a fresh install of Ubuntu Mate 16.04 (Ubiquity Robots, as recommended for Raspberry Pi by ROS Wiki) and MAVROS v 0.26.1.

I have previously had this running (couple of months ago) so something has been updated, perhaps.

mikeisted, 2018-08-14:
For further info, I am not using docker and have set up according to my blog (previously successfully!) at [https://mikeisted.wordpress.com/2018/05/13/robot-operating-system-for-flight-monitoring-and-control-getting-started/]

Voidminded, 2018-08-14:
There have been some changes in ROS recently, try:
```bash
$ sudo apt update
$ sudo apt upgrade
``` 
If ROS stuff needs to be upgraded, the upgrade should solve the issue.

vehsakul, 2018-08-23:
Same issure, I have used the script `ubuntu_sim_ros_gazebo.sh` from px4. Yesterday it finished successfully. Now I've needed to build mavros from source and got linker error with this symbol. Installed binaries, tried to launch them and here I am. Something has changed but what?

mikeisted, 2018-08-23:
OK - so update then upgrade worked for me.  Thanks Voidminded.

vehsakul, 2018-08-23:
@mikeisted indeed, I did `apt list --installed ros-kinetic* | grep upgradable | sed s#/.*## | xargs sudo apt install -y`

closed 2018-10-16 by TSC21