mavlink/mavros #1074: Setpoint_raw with bit mask 4067(VX VX PZ)not working properly.

eleboss, 2018-08-18:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
I am currently using setpoint_raw/local to control my drone indoor with a tfmini rangefinder. 
But when using setpoint_raw/local to control the VX VY and PZ with bitmask, strange things happen: 
I can switch to OFFBOARD mode and ARM the drone, however, the drone just stay on the ground with no movement and the propeller keep static(Usually, when ARMED, the propellers should rotate)

In the real world I done the following test:
control vx, vy, z, y -> 3043 (0xBE3):stay on the ground, propeller rotate
control vx, vy, z -> 4067 (0xFE3): stay on the ground, propeller keep static
control vx, vy, z, y_rate -> 2019 (0x7E3):stay on the ground, propeller rotate
control vx, vy, vz, y -> 3015 (0xBC7):working properly

and I also check basically all issues in google and github, but still can't figure out why this will happen.

In order the further confirm the problem, I try the simulation SILT in gazebo.with same code. and same thing happen, but this time I get more information,
control vx, vy, z, y -> 3043 (0xBE3):stay on the ground, propeller rotate with :WARN  [mavlink] force setpoint not supported
control vx, vy, z -> 4067 (0xFE3): stay on the ground, propeller keep static with WARN  [mavlink] force setpoint not supported
control vx, vy, z, y_rate -> 2019 (0x7E3):stay on the ground, propeller rotate with WARN  [mavlink] force setpoint not supported
control vx, vy, vz, y -> 3015 (0xBC7):working properly with no WARN

I am using the latest(1.8.0) version of PX4. From my viewpoint, It should work. and the [https://github.com/PX4/Firmware/issues/7068](url) comfim that. An I wrong about something?

BTW, many references mentioned that setpoint_raw/target_local is a loop back topic, while my setpoint_raw/target_local never return anything at anytime, It is normal?

## MY code

`
    position_pub = rospy.Publisher('mavros/setpoint_raw/local', PositionTarget, queue_size=1)
    stamp = rospy.get_rostime()

    raw_msg = PositionTarget(coordinate_frame=PositionTarget.FRAME_LOCAL_NED,
                             type_mask=
                                       PositionTarget.IGNORE_PX + PositionTarget.IGNORE_PY + PositionTarget.IGNORE_PZ +
                                       PositionTarget.IGNORE_AFX + PositionTarget.IGNORE_AFY + PositionTarget.IGNORE_AFZ +
                                        PositionTarget.IGNORE_YAW_RATE +PositionTarget.FORCE,

                             yaw = output_yaw)
    raw_msg.header.stamp = stamp
    raw_msg.velocity.x = 0.0
    raw_msg.velocity.z = 0.0                          
    raw_msg.position.z = 1
    position_pub.publish(raw_msg)
`
I use 
`rostopic echo /mavros/setpoint_raw/local `
to check the output, usually it looks like:
```
header: 
  seq: 128
  stamp: 
    secs: 1534585680
    nsecs: 509064912
  frame_id: ''
coordinate_frame: 1
type_mask: 4067
position: 
  x: 0.0
  y: 0.0
  z: 1.0
velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
[... 278 lines cut ...]
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "51"
      - 
        key: "Frequency (Hz)"
        value: "9.975074"
      - 
        key: "Last dt (ms)"
        value: "-0.002919"
      - 
        key: "Mean dt (ms)"
        value: "-0.000198"
      - 
        key: "Last system time (s)"
        value: "5.488630000"
      - 
        key: "Time offset (s)"
        value: "1534586337.964153528"
---

```

eleboss, 2018-08-18:
UPDATE:
 I accidently  make the setpoint_raw/target_local worked, when the drone is flying the data comes out!

AlexisTM, 2018-08-21:
The target_local topic is a feedback of the multicopter, not just a loopback.

The two masks I use are the following:

* MASK_POSITION         = 0b0000011111111000, //   x,   y,   z, vyaw
* MASK_VELOCITY         = 0b0000011111000111, // vx, vy, vz, vyaw

eleboss, 2018-08-21:
@AlexisTM 
Hi! 
Thanks for the very useful suggestions, I am currently using the velocity control, It works well, and the position control also work well in simulation environment.

But when use the combined mask, It just not working.

AlexisTM, 2018-08-21:
The implementation for combined commands is a bit a spaghetti in the backend (PX4 Firmware). 

NOTE: The Position setpoints in the PX4 Firmware are handled as a simple P velocity controller. I have good results with an offboard PD velocity controller.

eleboss, 2018-08-21:
@AlexisTM 
Yes maybe the spaghetti  implementation intorduced these bugs. Hope someone can solve that.

As you mentioned in the NOTE, the init parameter of Position setpoint actually not working well in my drone, My drone is a bit heavy. The drone swings heavely when use the setpoint_raw/local in position control. The RANGEFINDER[https://github.com/PX4/Firmware/issues/10264](url) bug may be the reason, I heven't done further test.

So, I implenent a cascade PID control by myself. First loop control the speed (setpoint_raw/local in speed mode), sencond loop control position, and with some pid tunning, It works well.

AlexisTM, 2018-08-21:
You mean you are sending one velocity setpoint then a position setpoint? The new setpoint erases the precedent one, more likely the position control is not aggressive enough compared to the velocity control. 

The Offboard mode is in rework, you can participate :)

eleboss, 2018-08-21:
I just using the setpoint velocity, fully give up position. the cascade is implement in my code.

AlexisTM, 2018-08-21:
Perfect ;)

OryWalker, 2018-08-24:
Sorry if I'm necroing, but I had a setpoint_raw position script that was working fine, switching the type mask to velocity only, and now it won't respond to any of the published velocity set_points, UNLESS I very specifically manually arm, initiate takeoff via mavcmd and then switch to offboard, all via terminal. It will then follow the velocity setpoints. Before, it arms, switches to offboard, and then sits on the ground until it disarms. 

Whereas with the other script (that only differs with respect to the type mask and the published setpoints being position rather than velocity) the drone would automatically take off. It would arm, switch to offboard, and then after a short wait would start following the setpoints. 

I'm at my wits end at this point and I have no idea what I need to do to my script to get it to consistently follow the velocity commands.

closed 2019-05-31 by TSC21

petertheprocess, 2020-10-16:
Is there any progress? I encountered the same problem when I use setpoint_raw/local to send a Vx Vy z cmd.

AlexisTM, 2020-10-16:
This was 2 year ago, PX4 developed a lot in between.

Take a look at the Mavlink receiver part, I believe it should be able to do Vx, Vy and Z commands.

https://github.com/PX4/Firmware/blob/7d6f8dc8821989cb3d6764a6d85fcb2d839809d4/src/modules/mavlink/mavlink_receiver.cpp#L784-L953