mavlink/mavros #1080: AP supports mocap_pose_estimate

rmackay9, 2018-09-07:
On the ArduPilot side we've added support for a number of the external position estimates mavlink messages including the [ATT_POS_MOCAP message](http://mavlink.org/messages/common#ATT_POS_MOCAP) ([AP code is here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L2230)) so I think we should remove "mocap_pos_estimate" from the plugin_blacklist in [apm_pluginlists.yaml](https://github.com/mavlink/mavros/blob/master/mavros/launch/apm_pluginlists.yaml#L11).

Here is an [example video](https://www.youtube.com/watch?v=P3qllsrt-_M) of it working on two Bebop2s (flashed with ArduPilot) performing an auto mission with a circle mission command.

vooon, 2018-09-07:
Great news, everyone!

closed 2018-09-07 by vooon

ashirsat, 2021-02-23:
Hi @vooon. 

Is this plugin support enabled for px4  firmware version 1.10.1 with mavros ?

commit 696674a4 (2018-09-07): launch: fix #1080: APM now support mocap messages