mavlink/mavros #1084: need clarification about frames

shabhu18, 2018-09-13:
I am testing offboard mode using setpoint position topic . if i have understood the documentation correctly the coordinate system followed is enu by mavros. So if i give a point say 5m in X direction  to my flying platform it should go by 5 m in east . 
My query is following 
Now i want my flying platform to go 5 m in front following REP 103 convention what transform should i do ?
1. Is there a transform already being published so i do just a transform lookup .?
2. If not what sensor links the enu to REP 103 convention (its it the orientation of iMU which gives me angle wrt to body frame coordinate following enu convention ) .? I am bit confused on this.

Any help is appreciated
thanks

This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ * ] PX4

Version: 3.8

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

FaboNo, 2018-10-09:
@shabhu18 May be you have found the solution by yourself already but the FCU is working with the NED frame and ROS is working with the ENU frame.
Mavros takes care of the transforms between the two frames.
So if you want to send a setpoint to the FCU, it must be defined in ENU frame (X forward and Z up), Mavros will do the transform in NED coordinates.
Conversely when you want to read the current pose of the quadrotor, you can  use mavros/local_position/pose, mavros will do the transformation so that you will read the pose in ENU as well
Hope it helps

Doodleshr, 2018-10-11:
@FaboNo 
Is ENU East-North-Up where x heads East, y heads north (forward) and z heads up?

Doodleshr, 2018-10-13:
@FaboNo thanks mate

closed 2019-05-31 by TSC21

RishabhChandrakar, 2026-02-21:
> [@shabhu18](https://github.com/shabhu18) May be you have found the solution by yourself already but the FCU is working with the NED frame and ROS is working with the ENU frame. Mavros takes care of [...]

@FaboNo  I have a doubt regarding this MAVROS frames.
The /mavros/local_position/pose topic gives position (x, y, z) in the ENU frame.
My understanding is that these coordinates are expressed in a fixed world (ENU) reference frame and are independent of the physical orientation of the flight controller.
So even if the flight controller is facing east, north, or any other direction, the (x, y, z) values should remain the same.
Only the orientation (yaw) should change when the drone rotates about the Z-axis.

Is this understanding correct?