mavlink/mavros #1087: ~/global_position/local outputs incorrect values

anuppari, 2018-09-19:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

The pose published on the `~/global_position/local` topic does not appear to be in UTM coordinates, as they should be according to the [ROS wiki](http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.global_position). Sample output:

```
header: 
  seq: 3748
  stamp: 
    secs: 1537318498
    nsecs:  34594432
  frame_id: "map"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: -2.21752996492
      y: 2.16391371669
      z: 0.064
    orientation: 
      x: -0.603754532149
      y: -0.0782669927127
      z: -0.783991483709
      w: -0.12129374098
  covariance: [42.497361000000005, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 42.497361000000005, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 18.173168999999998, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 99999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 99999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 99999.0]
twist: 
  twist: 
    linear: 
      x: 0.25
      y: 0.02
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```
whereas for my location, the UTM coordinates should be on the order of 300k and 3m for Easting and Northing.

By the way, the wiki seems incorrect compared to the actual default parameters. Here's part of the px4_config.yaml installed with mavros:

```
# global_position
global_position:
  frame_id: "map"             # origin frame
  child_frame_id: "base_link" # body-fixed frame
  rot_covariance: 99999.0   # covariance for attitude?
  gps_uere: 1.0             # User Equivalent Range Error (UERE) of GPS sensor (m)
  use_relative_alt: true    # use relative altitude for local coordinates
  tf:
    send: false               # send TF?
    frame_id: "map"  # TF frame_id
    global_frame_id: "earth"  # TF earth frame_id
    child_frame_id: "base_link" # TF child_frame_id
```
whereas the wiki lists completely different default frames, and list the tf send as defaulting to true. Also, what is the `global_frame_id`? Not listed on the wiki, and it's not clear what it's used for.

The tf tree also is disjoint. Is there some frame I should be setting up to create a connected tree?

[image]

### MAVROS version and platform

Mavros: Latest .deb
ROS: Kinetic
Ubuntu: 16.04
[... 418 lines cut ...]
        value: "1.922310"
      - 
        key: "Mean RTT (ms)"
        value: "0.781658"
      - 
        key: "Last remote time (s)"
        value: "3786.131563000"
      - 
        key: "Estimated time offset (s)"
        value: "1537315086.270706415"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 4225 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 33, 2, 36, 230, 1, 105, 74, 141, 111, 241, 242, 147, 245, 24, 4, 0, 30, 31

```

anuppari, 2018-09-19:
Also, the actual GPS coordinates are correct, as seen on the `~/global_position/global` topic

TSC21, 2018-09-19:
The wiki is outdated yes. It's still an effort that needs to be taken.

closed 2018-09-19 by TSC21

anuppari, 2018-09-19:
So how does one get UTM coordinates?

TSC21, 2018-09-19:
From MAVROS you currently don't. You can though use GeographicLib (a MAVROS dependency that's currently being used) and the available API from it to convert the data received on the global_position plugin to UTM.

anuppari, 2018-09-19:
So what is the data that is being published on `~/global_position/local`? Is that some local offset from a starting position? How is the starting position set? What about the starting orientation?

elgarbe, 2018-12-21:
I'm interested in this issue too. I was using global_position/local for UTM coordinates (on 0.19), but now in 0.27 I don't know what is global_position/local message is.
In 0.19 I have global position (UTM) and orientation. Now I need to look at two different topics to get the same data and add a coordinate conversion...

TSC21, 2018-12-21:
We do not support UTM coordinates. The local topic is basically the odometry of the vehicle with respect to the home position, which is the origin.

elgarbe, 2018-12-21:
is this https://github.com/mavlink/mavros/issues/1124 related to our concern?

TSC21, 2018-12-21:
> is this #1124 related to our concern?

Yes.

JaCoHSU, 2020-07-22:
I have an issue with mavros/global_position/local. I want to get the local position in gazebo but what I get is something strange:

---
header: 
  seq: 39903
  stamp: 
    secs: 801
    nsecs: 912005320
  frame_id: "map"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: 656765.748466
      y: 5113922.18278
      z: -0.268
    orientation: 
      x: 0.00969024305404
      y: -0.00910873705823
      z: 0.04495960274
      w: -0.998900329461
  covariance: [1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 99999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 99999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 99999.0]
twist: 
  twist: 
    linear: 
      x: -0.01
      y: -0.03
      z: 0.03
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---

Position x & y should be something like 0. Any idea what could cause this?

elgarbe, 2020-07-22:
If you need the odometry then you should use local_position/local.

JaCoHSU, 2020-07-22:
@elgarbe what is the difference between them?
If I spawn my model at x=5 and y=5 I would like to get these values as local position. Why am I getting such strange values ?

elgarbe, 2020-07-22:
There is 3 reference frames on this issue:
Gazebo: your x=5 and y=5. You can obtain that coordinates using model_state (or something like this). But that is useful in gazebo only, not in a real implementation. 
Px4 global position: When you spawn a model on gazebo the GPS on it is configured to be on some place in the world, so global position is referenced to that place. Gazebo x=0 and y=0 belong to that global position (I don't remember what that place is)
Px4 local position: it is the robot odometry, so it starts on x= and y=0 no matter where your model is in the world.
You could convert global_position/global to ECEF in order to get X/Y coordinate. Knowing what is ECEF coordinate of GPS home position, you can obtain your x=5, y=5.

take a look at this:
https://github.com/mavlink/mavros/issues/1245
https://github.com/mavlink/mavros/issues/1291
https://github.com/mavlink/mavros/issues/1213

JaCoHSU, 2020-07-22:
@elgarbe I am asking because I have used global_position/local before without any problems. But now on a different computer I got these strange values and I really don't know why. Did something change in Mavros lately? Is that a gazebo or a px4 problem? I really appreciate any kind of help :)

JaCoHSU, 2020-07-22:
@elgarbe thanks for that great answer. So what I am looking for is the px4 global position in local coordinates. So the x and y distance from the point you described with "some place in the world". I just need to know how I can change "that place in the world". I know that when I spawned the model at x=y=5 the global_position/local had x=y=5 and when I spawned a model at x=y=2 it was x=y=2. 

Any idea how I can set this reference point?

elgarbe, 2020-07-22:
there is not a simple answer for that. You will manage to transform different coordinates frames.
I don't think that px4 could ever know were you spawn your model in gazebo world.
Take a look at this test I made, maybe you could extract some ideas:
https://github.com/mavlink/mavros/issues/1279

JaCoHSU, 2020-07-22:
@elgarbe Hmm, I see ... 
When I start ros and show global_position/local I get x=y=0, after about 2 seconds it jumps to these high values. However, like you said if it is in ECEF the values are probably right but what I really dont get is that it worked before. 
Might geopraphic lib has anything to do with it? Or some other libs that arent installed correctly?

Like I said, it worked before on a different machine and is still working on a friends machine. Thats why I am still thinking that it is an installation or setup problem.

However, thank you very much for the fast responses and if you have any other ideas please let me know.

elgarbe, 2020-07-22:
Yes, mavros change from one version to another.
Could you check your friend and your mavros version?
Like you, I've some problems in the past with changes on new versions.