mavlink/mavros #1088: Using fake gps, no output at /mavros/global_postion/global

Doodleshr, 2018-09-27:
Hi guys, 

I am trying to use fake_gps plugin and I followed below steps:
1. set MAV_USEHILGPS to 1
2. Added fake_gps plugin to my px4_config.yaml.
3. publish vision 3Dof pose to /mavros/fake_gps/vision
4. I cannot set SYS_COMPANION to 257600 as I cannot find the parameter anywhere.

I tried to inspect the output of /mavros/global_position/global but nothing comes out. Am I doing it right? Do I have to set other parameters? I would really appreciate it if anyone can help me out.

Thanks

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version
PX4

Version: ?3.7.1?

### Node logs

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 35.05
 * /mavros/fake_gps/geo_origin/lat: 40.028907
 * /mavros/fake_gps/geo_origin/lon: 116.30301
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: False
 * /mavros/fake_gps/use_vision: True
 * /mavros/fcu_protocol: v2.0
[... 335 lines cut ...]
        value: "9.999918"
      - 
        key: "Last RTT (ms)"
        value: "0.328341"
      - 
        key: "Mean RTT (ms)"
        value: "0.305927"
      - 
        key: "Last remote time (s)"
        value: "455.480443000"
      - 
        key: "Estimated time offset (s)"
        value: "1538047720.923194647"
---

```

### Check ID

OK. I got messages from 1:1.

---
Received 6213 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 26, 30, 31, 32, 33, 74, 76, 87, 230, 105, 111, 241, 242, 245

TSC21, 2018-09-27:
Ẁhat config did you set for `fake_gps` in `px4_config.yaml`?

Doodleshr, 2018-09-28:
@TSC21 

Thanks for your replay, I've updated my desciption. 

Cheers
HR

Doodleshr, 2018-09-30:
@TSC21 

I got reponse from /mavros/global_postion/global after several restarts but the lat and long value are still Zurich default value rather than my modified lat and long. 

The output of /mavros/fake_gps/vision has the exact same value as what I published to it; 
while the output of /mavros/global_postion/global is : 
```
header: 
  seq: 303
  stamp: 
    secs: 1538300528
    nsecs: 145340672
  frame_id: "base_link"
status: 
  status: 0
  service: 1
latitude: 47.3977608
longitude: 8.5456112
altitude: 545.260900428
position_covariance: [4.0, 0.0, 0.0, 0.0, 4.0, 0.0, 0.0, 0.0, 4.0]
position_covariance_type: 2
```

although I set what follows in px4_config.yaml
```
  use_mocap: false         # ~mocap/pose
  mocap_transform: false   # ~mocap/tf instead of pose
  use_vision: true       # ~vision (pose)
  geo_origin:
    lat: 40.028907        # latitude [degrees]
    lon: 116.30301           # longitude [degrees]
    alt: 35.05            # altitude (height over the WGS-84 ellipsoid) [meters]
  eph: 2.0
  epv: 2.0
```

the output of /mavros/global_position/local  and /mavros/local_position/pose are similar and are roughly zero:
```header: 
  seq: 24455
  stamp: 
    secs: 1538301993
    nsecs: 959581248
  frame_id: "map"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: 0.0528472390014
      y: 0.0111187947004
      z: 0.081
    orientation: 
      x: 0.0214311271929
      y: -0.00541117329422
      z: -0.515657798546
      w: -0.856509477697
  covariance: [4.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 4.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 4.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 99999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 99999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 99999.0]
```

the output of /mavros/fake_gps/vision is some arbitary constant value I set: 
```
header: 
  seq: 0
  stamp: 1538302172.676561887
  frame_id: 
pose: 
  position: 
    x: 10
    y: 21
    z: 12.5
  orientation: 
    x: 0
    y: 0
    z: 0
    w: 0
```

I am testing above functions in the gazebo simulator but I dont think it has anything to do with the result.

Thanks again for your help.
HR

TSC21, 2018-09-30:
You are using SITL right? Probably you are not aware, but Gazebo publishes GPS data to the flight controller in SITL mode.

Doodleshr, 2018-10-01:
thanks, closing the issue.

closed 2018-10-01 by Doodleshr

Coderx7, 2022-04-11:
hi 
@TSC21  is there a way to somehow get this to work in SITL? I have been trying for couple of days now, but no matter what I do, I cant get this to work?