mavlink/mavros #1089: Controlling a gimbal via mavros

jannsta1, 2018-09-27:
### Issue details

I'm trying to figure out the best way to control an onboard gimbal through mavros. Since I'm using PX4 I think that the recommended way to do this is via the ```actuator_control``` plugin?

In my initial setup, rqt_graph confirms that the mavros node is subscribing to the "/mavros/actuator_control" topic that I'm sending. I'm sending all 8 "control" bytes as some value between 0 and 1, the group_mix is set to 2 and I'm also definitely transitioning to offboard mode. However, looking at the mavlink inspector (on QGroundControl) I don't see the ACTUATOR_CONTROL_TARGET message that it looks like the mavros actuator_control plugin publishes on. 

Since its mentioned a lot in these issues I've tried using wireshark to look at the data between ports 14540 -> 14557  but so far I'm finding it quite tricky to interpret the data (is there any documentation on how its best to setup filters for mavlink? or is there a script commonly used for this - [this](https://github.com/ArduPilot/pymavlink/blob/19473add78c0d6fd5b323aded5eaabf2d60446dd/generator/mavgen_wlua.py) for example)

Any comments on how I can verify mavros is publishing the expected message, or comments on my general assumptions, would be appreciated.

### MAVROS version and platform

Mavros: 0.26.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: 1.8.0

### Node logs

```
[ INFO] [1538072468.952158126]: FCU URL: udp://:14540@localhost:14557
[ INFO] [1538072468.955056961]: udp0: Bind address: 0.0.0.0:14540
[ INFO] [1538072468.955238033]: udp0: Remote address: 127.0.0.1:14557
[ INFO] [1538072468.955573715]: GCS bridge disabled
[ INFO] [1538072469.000820102]: Plugin 3dr_radio loaded
[ INFO] [1538072469.007234190]: Plugin 3dr_radio initialized
[ INFO] [1538072469.007520081]: Plugin actuator_control loaded
[ INFO] [1538072469.015664651]: Plugin actuator_control initialized
[ INFO] [1538072469.024465423]: Plugin adsb loaded
[ INFO] [1538072469.033386170]: Plugin adsb initialized
[ INFO] [1538072469.033835454]: Plugin altitude loaded
[ INFO] [1538072469.035453578]: Plugin altitude initialized
[ INFO] [1538072469.035688804]: Plugin cam_imu_sync loaded
[ INFO] [1538072469.041852582]: Plugin cam_imu_sync initialized
[ INFO] [1538072469.042148465]: Plugin command loaded
[ INFO] [1538072469.049401281]: Plugin command initialized
[ INFO] [1538072469.049642449]: Plugin debug_value loaded
process[fake_vicon_wasp_1-9]: started with pid [13437]
[ INFO] [1538072469.066907375]: Plugin debug_value initialized
[ INFO] [1538072469.067044199]: Plugin distance_sensor blacklisted
[ INFO] [1538072469.067316047]: Plugin fake_gps loaded
process[image_view-10]: started with pid [13440]
[ INFO] [1538072469.107123853]: Plugin fake_gps initialized
[ INFO] [1538072469.107425965]: Plugin ftp loaded
[ INFO] [1538072469.166031544]: Plugin ftp initialized
[ INFO] [1538072469.166266287]: Plugin global_position loaded
[ INFO] [1538072469.216682624]: Plugin global_position initialized
[ INFO] [1538072469.217004270]: Plugin gps_rtk loaded
[ INFO] [1538072469.236738535]: Plugin gps_rtk initialized
[ INFO] [1538072469.236968690]: Plugin hil loaded
[ INFO] [1538072469.265049071]: Plugin hil initialized
[ INFO] [1538072469.265269847]: Plugin home_position loaded
[ INFO] [1538072469.271734776]: Plugin home_position initialized
[ INFO] [1538072469.271912252]: Plugin imu loaded
[ INFO] [1538072469.289253806]: Plugin imu initialized
[ INFO] [1538072469.289475499]: Plugin local_position loaded
[ INFO] [1538072469.299682054]: Plugin local_position initialized
[ INFO] [1538072469.299882717]: Plugin log_transfer loaded
[ INFO] [1538072469.305046473]: Plugin log_transfer initialized
[ INFO] [1538072469.305230431]: Plugin manual_control loaded
[ INFO] [1538072469.309929692]: Plugin manual_control initialized
[ INFO] [1538072469.310102952]: Plugin mocap_pose_estimate loaded
[ INFO] [1538072469.317420976]: Plugin mocap_pose_estimate initialized
[ INFO] [1538072469.317638121]: Plugin obstacle_distance loaded
[... 214 lines cut ...]
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.198645"
      - 
        key: "Last remote time (s)"
        value: "42.665364000"
      - 
        key: "Estimated time offset (s)"
        value: "2.762011076"

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 6008 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 265, 140, 141, 147, 24, 26, 30, 31, 32, 36, 74, 76, 77, 83, 85, 87, 102, 230, 105, 111, 241, 245

```

vooon, 2018-09-28:
I don't sure that FW respond back with that message.
And perhaps you should setup mixer to be able to use that message's data.

jannsta1, 2018-10-05:
It looks like reception of SET_ACTUATOR_CONTROL_TARGET functionality was added recently in [this](https://github.com/PX4/Firmware/pull/7477) PX4 pull request. 

Also its important to ensure vmount isn't running (```vmount stop``` can be added to the rcS file on startup) otherwise there is a conflict between the 2 controllers. Not sure how this works on a real drone yet.

closed 2018-10-05 by jannsta1