mavlink/mavros #1090: receiving px4flow images in mavros_extras

chataign, 2018-10-04:
I am trying to modify mavros_extras and px4/Flow in order to optionally send the px4flow images and receive them in the mavros_extras px4flow plugin so they can be recorded in a rosbag for future debugging (eg. to have a record of areas where the flow quality is low). I know that images can be seen with the USB cable connected, but it is not very practical for my tests.

I modified px4/Flow to send the images even without USB, and to publish them on MAVLINK_COMM_0 instead of MAVLINK_COMM_2.

Then I handled the Mavlink handshake and encapsulated data messages in mavros_extras/px4flow similarly to the px4flow node (http://wiki.ros.org/px4flow_node), see code below. However I am not receiving any handshakes or encapsulated data in the plugin.

Is there a reason why this is not working?  

```
void handle_data_transmission_handshake(const mavlink_message_t *msg, uint8_t sysid, uint8_t compid) 
{
       mavlink_msg_data_transmission_handshake_decode( msg, &handshake );

    ROS_INFO_ONCE( "received handshake, size=%d %dx%d packets=%d", 
        (int)handshake.size, 
        (int)handshake.width, 
        (int)handshake.height, 
        (int)handshake.packets );

        if ( image_buffer.size() < handshake.size )
            image_buffer.resize( handshake.size );
}

void handle_encapsulated_data(const mavlink_message_t *msg, uint8_t sysid, uint8_t compid) 
{
       if ( handshake.size == 0 || handshake.packets == 0)
           return;

        mavlink_encapsulated_data_t img;
        mavlink_msg_encapsulated_data_decode( msg, &img );

    ROS_INFO_ONCE( "received image=%d", (int)img.seqnr ); 

        size_t seq = img.seqnr;
        size_t pos = seq * handshake.payload;

        if ( seq+1 > handshake.packets )
            return;

        size_t bytesToCopy = handshake.payload;

        if ( pos + handshake.payload >= handshake.size )
             bytesToCopy = handshake.size - pos;

        memcpy( &image_buffer[pos], img.data, bytesToCopy );

        if (  seq+1 == handshake.packets)
        {
            sensor_msgs::Image image;
            image.header.frame_id = frame_id;
            image.height = handshake.height;
            image.width = handshake.width;
            image.encoding = sensor_msgs::image_encodings::MONO8;
            image.is_bigendian = false;
            image.step = handshake.width;

            image.data.resize(handshake.size);
            memcpy( &image.data[0], &image_buffer[0], handshake.size );

            image_pub.publish(image);
        }
} 
```

vooon, 2018-10-04:
You can try old version of mavros where that is implemented: https://github.com/mavlink/mavros/blob/0.17.5/mavros_extras/src/plugins/image_pub.cpp

chataign, 2018-10-05:
Thank you, i will compare this with my code.

However some changes are also required on the firmware side, correct? So that images are sent even if USB isn't connected and on a different com port?

Thanks

chataign, 2018-10-05:
I tried your code and it works great when connected to the px4flow via USB

Is it possible to send the images, via the Pixhawk, to the companion computer on the MAV ?

TSC21, 2018-10-05:
@chataign no. The flight controller does not receive or forward that data.

closed 2018-10-16 by TSC21