mavlink/mavros #1091: mavlink/actuator_control

FaboNo, 2018-10-08:
If we want to control a quadcopter so that it follows a polynomial trajectory, is it possible to use mavlink/actuator_control topic ?

If so:
1. We need to set group_mix =  PX4_MIX_FLIGHT_CONTROL,
2. We have to fill in the first 4 values in controls (roll, pitch, yaw, throttle),
3. We need to be OFFBOARD mode as well.

Am I correct?

Thank you
----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x] PX4

Jaeyoung-Lim, 2018-10-09:
@FaboNo It is a better approach to use angular velocity commands rather than actuator control topics.

If you use the actuator_control topic, there is no feedback and the offboard setpoints are limited with the bandwidth with your connection to the flight controller.

If you use the angular velocity topics, the control loop of the autopilot will control the vehicle to follow the setpoints that you have provided to the flight controller.

FaboNo, 2018-10-09:
@Jaeyoung-Lim Thank you for your message 
If I follow you it is better to use setpoint_velocity and send the roll, pitch and yaw angulaur velocity only (i mean no linear velocity and no x,y,z position), am I correct? Is it what you are doing at ETH to implement the polynomial trajectories?

Jaeyoung-Lim, 2018-10-10:
@FaboNo I use a geometric controller to make the quad follow polynomial trajectories which the quad is given differentially flat references.

I have a WIP implementation on a geometric controller [here](https://github.com/Jaeyoung-Lim/mavros_controllers) if you want to look into.

FaboNo, 2018-10-10:
@Jaeyoung-Lim Thank you for the link and I think I will have to read the article from Fassler & all in order to understand your code. Looks great when I look at the deom with gazebo:)