mavlink/mavros #1094: Better communication between my computer and my drone / Change in Mavros/Mavlink settings

shortcut21, 2018-10-12:
Hello, guys! Can you please give me a solution or some advice on my issue.

### Issue details

I want to achieve a good communication between my drone and my Linux computer. I have Pixhawk 3 Pro and just now I got Laird Tech RM024 P10 M 30- 2.4GHz module, since my previous radio (3DR Radio from Drotek 433MHz) was apparently too slow. Unfortunately, I am unable to connect correctly my system because I keep getting ¨RTT too high..." and ¨PR: request param #¨. At first I set the radios with 115200 baud but when I ran the command: roslaunch mavros px4.launch it stoped after Mavros Started My ID 1240 Target Id 1.1 and it didn´t run at all.
However, when I set the radio to be with baud 57600 it connects but I keep getting the previous {WARN}-ings 
I suppose it has to do with settings in some of the main files but can you please give me some advice on how to connect them and have 0 warnings.  
How can I use the best of my components and how to set them correctly? 
I would appreciate each comment given. 

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04 LTS

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: ?3.7.1?

### Node logs
rosrun mavros mavros_node _fcu_url:=/dev/ttyUSB0:57600
[ INFO] [1539358397.728280382]: FCU URL: /dev/ttyUSB0:57600
[ INFO] [1539358397.729885050]: serial0: device: /dev/ttyUSB0 @ 57600 bps
[ INFO] [1539358397.731532982]: GCS URL: udp://@
[ INFO] [1539358397.731843713]: udp1: Bind address: 0.0.0.0:14555
[ INFO] [1539358397.741842662]: Plugin 3dr_radio loaded
[ INFO] [1539358397.743551270]: Plugin 3dr_radio initialized
[ INFO] [1539358397.743643832]: Plugin actuator_control loaded
[ INFO] [1539358397.746644659]: Plugin actuator_control initialized
[ INFO] [1539358397.749251627]: Plugin adsb loaded
[ INFO] [1539358397.752438647]: Plugin adsb initialized
[ INFO] [1539358397.752559438]: Plugin altitude loaded
[ INFO] [1539358397.753471971]: Plugin altitude initialized
[ INFO] [1539358397.753570069]: Plugin cam_imu_sync loaded
[ INFO] [1539358397.754187962]: Plugin cam_imu_sync initialized
[ INFO] [1539358397.754322662]: Plugin command loaded
[ INFO] [1539358397.758884624]: Plugin command initialized
[ INFO] [1539358397.759005763]: Plugin debug_value loaded
[ INFO] [1539358397.763317878]: Plugin debug_value initialized
[ INFO] [1539358397.763449717]: Plugin distance_sensor loaded
[ WARN] [1539358397.764278225]: DS: plugin not configured!
[ INFO] [1539358397.764322614]: Plugin distance_sensor initialized
[ INFO] [1539358397.764465354]: Plugin fake_gps loaded
[ INFO] [1539358397.772476839]: Plugin fake_gps initialized
[ INFO] [1539358397.772671338]: Plugin ftp loaded
[ INFO] [1539358397.780678090]: Plugin ftp initialized
[ INFO] [1539358397.780878268]: Plugin global_position loaded
[ INFO] [1539358397.795481575]: Plugin global_position initialized
[ INFO] [1539358397.795649452]: Plugin gps_rtk loaded
[ INFO] [1539358397.798041967]: Plugin gps_rtk initialized
[ INFO] [1539358397.798199089]: Plugin hil loaded
[ INFO] [1539358397.811797552]: Plugin hil initialized
[ INFO] [1539358397.811935460]: Plugin home_position loaded
[ INFO] [1539358397.815300937]: Plugin home_position initialized
[ INFO] [1539358397.815451356]: Plugin imu loaded
[ INFO] [1539358397.821749766]: Plugin imu initialized
[ INFO] [1539358397.821882983]: Plugin local_position loaded
[ INFO] [1539358397.825387745]: Plugin local_position initialized
[ INFO] [1539358397.825501137]: Plugin log_transfer loaded
[ INFO] [1539358397.828277751]: Plugin log_transfer initialized
[ INFO] [1539358397.828398538]: Plugin manual_control loaded
[ INFO] [1539358397.831269255]: Plugin manual_control initialized
[ INFO] [1539358397.831371863]: Plugin mocap_pose_estimate loaded
[ INFO] [1539358397.834266785]: Plugin mocap_pose_estimate initialized
[ INFO] [1539358397.834371593]: Plugin obstacle_distance loaded
[ INFO] [1539358397.836563644]: Plugin obstacle_distance initialized
[... 195 lines cut ...]
    message: "Low voltage"
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Voltage"
        value: "9.80"
      - 
        key: "Current"
        value: "0.4"
      - 
        key: "Remaining"
        value: "0.0"
---

### Check ID

WARNING: mavros/target_system_id not set. Used default value: 1
WARNING: mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
OK. I got messages from 1:1.

---
Received 835 messages, from 1 addresses
sys:comp   list of messages
  1:1     147, 1, 36, 230, 32, 105, 74, 44, 141, 77, 241, 83, 245, 22, 4, 30, 0

vooon, 2018-10-14:
Try to lower rate or disable time synchronise module.

shortcut21, 2018-10-15:
Hello, @vooon! Thank you for your fast reply. I disabled the the Network time and NTP synchronized but I still get the RTT too high [WARN]ing. Is there a way to change the settings to adapt the high rate instead of changing it to lower. I really want to achieve the best performance of this telemetry module.

vooon, 2018-10-15:
I mean that you should change https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml#L12 or blacklist sys_time plugin.

shortcut21, 2018-10-15:
Oh, ok. 
I did it just now, but I get the _PR: request param #_ WARN what am I missing? Or what might be the problem?

vooon, 2018-10-15:
Param timeout almost normal, at least until retries left > 0.
Basically that means, that some of param <id> is lost in transmission and mavros will retry to get that parameter value.
Param plugin gives 3 retries for each parameter, but that process is slow.
So bacially better to give some time for initial comms like param and mission download.

shortcut21, 2018-10-16:
Thank you for your help, @vooon ! 
However, I have couple more questions. 
1. So there is no best case scenario when there are no WARN-ings? 

2. What is more, today I tested the connection by ros launching the px4.launch file and in 40sec the parameters loaded. I decided to configure the radio to higher baud rate: 115200 and try again. I got radio failsafe light on my pixhawk. (I switched back to 57600)  Why is higher baud not possible to be connected? 

3. I was using two computers connected via cable. Today when I ran the tests, my ground station (that is on one of the computers) was switching the battery levels between the real battery and the one of the pixhawk every 2 seconds. So I decided to use only one computer and installed QGC on the ROS Mavros computer and ran everything there. Everything is well connected but initially I get 3 ERRORS 
FCU: IGN REQUEST LIST: Busy
FCU: Operation timeout

I am currently trying different approaches and solutions online.
Thanks again for your support!

vooon, 2018-10-16:
40 sec is a quite too long, should be done around 10-20 secs. But if you have lost some of params each one gives additional second.
Better if you don't have any warnings, but at least this ones is autorecoverable.

When you change baud rate do you change speed parameters too?

shortcut21, 2018-10-17:
Whenever I change the baud rate I change also the fcu_url in the px4.launch file and when I run it stops at _MAVROS started My ID 1240_. Should I also change something in the congif file?
I have quite some missing parameters. What is the reason? How do you achieve to have no warnings?

TSC21, 2018-10-17:
@shortcut21 how are yoi configuring the baud rate on the PX4 side? Are you using PX4 upstream master? What's the last commit tag of it in that case?

shortcut21, 2018-10-17:
Hello, @TSC21 ! 
I haven't touched anything but the px4.launch file. Please give me guidance how to set it on the PX4 side. I read some comments on other issues, but I couldn't find it. On the SD card in the microsd folder I don't have /etc.
Should I change something in the px4fmu_common/ init.d/rcS or somewhere else and how? Where can I get more information about how to make my params so I don't get any warnings?

TSC21, 2018-10-17:
Without knowing what version of PX4 you are using, that would be hard, because the current approach on upstream master is different from the one on the stable version. If you are using the upstream master, then I advise you to read and follow https://docs.px4.io/en/peripherals/serial_configuration.html and https://docs.px4.io/en/peripherals/mavlink_peripherals.html. `SYS_COMPANION` parameter doesn't exist anymore (https://github.com/PX4/Devguide/pull/621)

shortcut21, 2018-10-18:
I am using the v1.8.0 Stable Release. I haven't changed anything on the files in the Firmware yet. What would be the best approach then? And where (in which terminal) should I change this parameters?

On the link : https://docs.px4.io/en/peripherals/serial_configuration.html 
> How to Configure a Port
All the serial drivers/ports are configured in the same way:
Set the configuration parameter for the service/peripheral to the port it will use.
Reboot the vehicle.
Set the baud rate parameter for the selected port to the desired value.
Configure module-specific parameters (i.e. MAVLink streams and data rate configuration).
Configuration parameter names follow the pattern \*\_CONFIG or \*\_CFG. QGroundControl only displays the parameters for services/drivers that are present in firmware.

Where and how to change that?
I am glad that you guys replied :)

TSC21, 2018-10-18:
If you are using 1.8.0 stable then that guide is not helpful to you. You need to change the `SYS_COMPANION` parameter I told you about so to match the baud rate you have configured on MAVROS.

TSC21, 2018-10-18:
Also you should increase the baud rate on both sides.

shortcut21, 2018-10-18:
The desired baud rate is 115200. There is no such option in QGrounControl. I set teh SYS_COMPANION to Companion Link ( 921600) but when I reboot it says SYS_COMPANION Disabled.

shortcut21, 2018-10-18:
The desired baud rate is 115200. There is no such option in QGrounControl. I set the SYS_COMPANION to Companion Link ( 921600) but still the run of the px4.launch file stops at "MAVROS started MY ID 1.240..."

TSC21, 2018-10-18:
How are you expecting it to work if you have the baud rate set to 115200 on MAVROS and then set it to 921600 on the PX4 side?

shortcut21, 2018-10-18:
I know it doesn't make sense but I read it in a discussion . The maximum interface baud I can set on the radio is 230400 and qgc doesn't support that either. Should I go back to 57600 then?

shortcut21, 2018-10-22:
Please give me a hint on how to avoid missing parameters.

TSC21, 2018-10-22:
At that baud rate what I advise you is to lower the Mavlink stream rate of some of the messages or even remove the stream of some. That has to be done on the PX4 side.

closed 2018-10-22 by TSC21

shortcut21, 2018-10-24:
I can see the MAVLink messages in the MAVLink inspection in QGroundControl. Can you tell me exactly where can I remove the stream of others that I don't use, in which file or console should I change?

scd5027, 2019-05-22:
I don't think this will help you with the "RTT too high" warnings but if you want to use a different baud rate you'd have to change the configuration of the radios to operate using the requested baud rate.  If you are using SiK radios you can likely configure them with "AT" commands or using a SiK radio config tool.