mavlink/mavros #1107: Heartbeats from multiple component IDs handled incorrectly

edwinhayes, 2018-10-26:
### Issue details

On a system which has multiple MAVLink components, MAVROS will see heartbeat messages from multiple components, only one of which will be the flight controller (probably).  These might not all have the same status.  At present, the topic `/mavros/state` seems to get published repeatedly for the different components, but the topic has no field to differentiate different components, so the published state might: a - fluctuate between one or more states, b - not be representative of the flight controller state.

I haven't investigated, but I assume this same thing applies to other topics as well?  What happens to a topic publishing attitude if you have both a flight controller and gimbal publishing attitude information?

There is a parameter `/mavros/target_component_id` but I'm not sure what that actually does?  I feel like it should be filtering the handling of incoming messages so that topics which are supposed to represent the state of the vehicle are only using flight controller messages.

### MAVROS version and platform

Mavros: 0.26.3?
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

Version: Currently 3.6-rc12, but other versions too.

vooon, 2018-10-26:
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/sys_status.cpp#L644

HB only checks system id, and i don't remember why...

TSC21, 2019-05-05:
Fixed in #1228.

closed 2019-05-05 by TSC21