mavlink/mavros #1108: How to check if client is connected?

kossumov, 2018-10-28:
Hey guys
How can I check if clinet is connected? I have to do certain things if client has been disconnected

Thanks

vooon, 2018-11-06:
For gcs connection status not checked. Only possible way to achieve that - modify gcs_link "message received callback" to add timeout timer.
Then you can detect disconnects.

kossumov, 2018-12-20:
Thanks. Where in the code can I do it?

kossumov, 2018-12-21:
Basically, What I was trying to accomplish is to disarm pixhawk if gcs disconnects.
Seems like mavros node turns off every time GCS unexpectedly disconnects. (By "unexpectedly" I mean if user does not click "disconnect" and closes Mission Planner, or internet turns off")
Is there a way to disarm pixhawk when GCS disconnects. (or when mavros turns off)?

vooon, 2018-12-24:
But you don't even need to change anything. FCU already can react (or toggle failsafe) on that. Only what needed to be clarified - does they differentiate OBC and GCS heartbeats.

closed 2019-05-31 by TSC21