mavlink/mavros #1109: Sending SLAM coordinates to Pixhawk

trishantroy, 2018-10-29:
Hello,
I am using a Pixhawk with the latest stable PX4 Firmware. I want to use setpoint functions to manuever the quad to go from one point to another. The environment is GPS-denied. So, I am not using a GPS module. I have implemented SLAM to get current coordinates of the quad in x,y,z format. I want to use this data for setpoint functions. Is there a direct rostopic where these values have to be published? 
From what I understood, fake gps is a feature in PX4 firmware that is used to do this. But, I don't know how to implement this. Is there a documentation of the instructions to implement this?

Thank you in advance.

### MAVROS version and platform

Mavros: ?0.26.3?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] PX4

### Node logs

```
trishant@Eurus-Holmes:~/mavros_ws$ roslaunch mavros px4.launch 
... logging to /home/trishant/.ros/log/fc7911be-dbbc-11e8-81ba-2c6e8522ef61/roslaunch-Eurus-Holmes-5725.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Eurus-Holmes:35409/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
[... 353 lines cut ...]
      - 
        key: "Mean RTT (ms)"
        value: "1.200836"
      - 
        key: "Last remote time (s)"
        value: "97.415648000"
      - 
        key: "Estimated time offset (s)"
        value: "1540846524.276278496"
---

```

### Check ID

```
trishant@Eurus-Holmes:~/mavros_ws$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 4877 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 0, 2, 245, 4, 230, 1, 105, 74, 140, 141, 77, 111, 241, 83, 147, 36, 30, 31

```

TSC21, 2018-10-29:
>  I have implemented SLAM to get current coordinates of the quad in x,y,z format. I want to use this data for setpoint functions.

What exactly are you trying to do? Use the position feedback so to correct it using setpoints? If you are using PX4, then you want to use the Offboard mode, and its usage has been documented for a while now:
- https://docs.px4.io/en/flight_modes/offboard.html
- https://dev.px4.io/en/ros/mavros_offboard.html

closed 2018-10-29 by TSC21

trishantroy, 2018-11-02:
> What exactly are you trying to do? Use the position feedback so to correct it using setpoints?

Yes, I want to use the SLAM coordinates to tell Pixhawk what my current position is? Isn't this required to use the setpoint code shown in (https://dev.px4.io/en/ros/mavros_offboard.html)? Is the data from Pixhawk's internal sensors sufficient to achieve setpoint control? 
I thought that this data wouldn't be so accurate so wanted to use SLAM.

trishantroy, 2018-11-02:
> But you wouldn't need a position source if you were actually using attitude setpoints instead.

I did not understand this part. Setpoints would require a feedback who will tell the current position, right?

TSC21, 2018-11-02:
@trishantroy position setpoints is a type of setpoint. You also have velocity and attitude. For the attitude, you need the vehicle attitude feedback, not position. You can also implement a controller that sends attitude setpoints and controls the vehicle position based on the vehicle position feedback. There are a lot of ways of doing this. But just focus on using position setpoints then, that would be easy for you to implement.

trishantroy, 2018-11-02:
> But just focus on using position setpoints then, that would be easy for you to implement.

Yes, I want to implement the position setpoints feature. To do that, the Pixhawk would require the current position data from somewhere. 
What should I do so that the Pixhawk realizes it's position from the SLAM data?

TSC21, 2018-11-02:
@trishantroy https://dev.px4.io/en/ros/external_position_estimation.html.

Let's close this issue now. Just bring your questions to PX4 Slack or the PX4 forums.