mavlink/mavros #1110: I come across a assertion here

GaoGeolone, 2018-10-31:
https://github.com/mavlink/mavros/blob/1e19a1b8d73e6c380ca31f84a12c6036313cb7de/mavros/src/plugins/hil.cpp#L337

When I try to make a project on my odroid installed with ubuntu18.04 mate and ROS melodic, all the compiling work complete with some note, but nothing serious. but when I try to start my mavros_node, is will turn out nearly one time per three times .
here is my gdb result
`
[ INFO] [1540970736.550125755]: Plugin gps_rtk loaded
[ INFO] [1540970736.555932602]: Plugin gps_rtk initialized
mavros_node: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.

Thread 1 "mavros_node" received signal SIGABRT, Aborted.
__libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
47	../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S: No such file or directory.
(gdb) bt
#0  __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
#1  0xb6841b32 in __libc_signal_restore_set (set=0xbeffd514) at ../sysdeps/unix/sysv/linux/nptl-signals.h:80
#2  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:48
#3  0xb684282e in __GI_abort () at abort.c:79
#4  0xb683cd22 in __assert_fail_base (fmt=0xb68f2a78 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=0xb6f8cbf0 "(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.htm"..., assertion@entry=0x0, 
    file=0xb6f8cbbc "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", file@entry=0xb4021010 "\001", line=109, line@entry=3069758396, 
    function=function@entry=0xb6f8d870 <Eigen::internal::plain_array<double, 4, 0, 16>::plain_array()::__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]") at assert.c:92
#5  0xb683cdaa in __GI___assert_fail (assertion=0x0, file=0xb4021010 "\001", line=3069758396, 
    function=0xb6f8d870 <Eigen::internal::plain_array<double, 4, 0, 16>::plain_array()::__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]") at assert.c:101
#6  0xb6f25050 in Eigen::internal::plain_array<double, 4, 0, 16>::plain_array() ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#7  0xb6f24eda in Eigen::DenseStorage<double, 4, 4, 1, 0>::DenseStorage() ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#8  0xb6f24cd8 in Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >::PlainObjectBase() ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#9  0xb6f2840e in Eigen::Matrix<double, 4, 1, 0, 4, 1>::Matrix() () from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#10 0xaf250820 in Eigen::Quaternion<double, 0>::Quaternion() ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib//libmavros_plugins.so
**#11 0xaf35c114 in mavros::std_plugins::HilPlugin::HilPlugin() ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib//libmavros_plugins.so**
#12 0xaf36e1c8 in class_loader::impl::MetaObject<mavros::std_plugins::HilPlugin, mavros::plugin::PluginBase>::create() const ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib//libmavros_plugins.so
#13 0xb6f5ec14 in mavros::plugin::PluginBase* class_loader::impl::createInstance<mavros::plugin::PluginBase>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, class_loader::ClassLoader*) ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#14 0xb6f5c04c in mavros::plugin::PluginBase* class_loader::ClassLoader::createRawInstance<mavros::plugin::PluginBase>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#15 0xb6f5895e in boost::shared_ptr<mavros::plugin::PluginBase> class_loader::ClassLoader::createInstance<mavros::plugin::PluginBase>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
---Type <return> to continue, or q <return> to quit---
#16 0xb6f53822 in boost::shared_ptr<mavros::plugin::PluginBase> class_loader::MultiLibraryClassLoader::createInstance<mavros::plugin::PluginBase>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#17 0xb6f4f9d4 in pluginlib::ClassLoader<mavros::plugin::PluginBase>::createInstance(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#18 0xb6f4786c in mavros::MavRos::add_plugin(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&) ()
   from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#19 0xb6f45eac in mavros::MavRos::MavRos() () from /home/odroid/Odroid_Scout_ROS/devel_isolated/mavros/lib/libmavros.so
#20 0x004438b8 in main ()

`
and as what I have learned from the [http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.htm](url)
I think this may caused by my 32bits odroid

GaoGeolone, 2018-10-31:
I solve this issue by adding `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` here 
`class  HilPlugin : public plugin::PluginBase {`
`public:`
`	HilPlugin() : PluginBase(),`
`		hil_nh("~hil")`
`	{ }`
`	**EIGEN_MAKE_ALIGNED_OPERATOR_NEW**`
`	void initialize(UAS &uas_)`
`	{`
`		PluginBase::initialize(uas_);`

`		hil_state_quaternion_sub = hil_nh.subscribe("state", 10, &HilPlugin::state_quat_cb, this);`
`		hil_gps_sub = hil_nh.subscribe("gps", 10, &HilPlugin::gps_cb, this);`
`		hil_sensor_sub = hil_nh.subscribe("imu_ned", 10, &HilPlugin::sensor_cb, this);`
`		hil_flow_sub = hil_nh.subscribe("optical_flow", 10, &HilPlugin::optical_flow_cb, this);`
`		hil_rcin_sub = hil_nh.subscribe("rc_inputs", 10, &HilPlugin::rcin_raw_cb, this);`

`		hil_controls_pub = hil_nh.advertise<mavros_msgs::HilControls>("controls", 10);`
`		hil_actuator_controls_pub = hil_nh.advertise<mavros_msgs::HilActuatorControls>("actuator_controls", 10);`
`	}`

`	Subscriptions get_subscriptions()`
`	{`
`		return {`
`			       make_handler(&HilPlugin::handle_hil_controls),`
`			       make_handler(&HilPlugin::handle_hil_actuator_controls),`
`		};`
`	}`

GaoGeolone, 2018-10-31:
class HilPlugin : public plugin::PluginBase {
public:
	HilPlugin() : PluginBase(),
		hil_nh("~hil")
	{ }
	**EIGEN_MAKE_ALIGNED_OPERATOR_NEW**
	void initialize(UAS &uas_)

GaoGeolone, 2018-10-31:
@TSC21 
My test turns out to be normal,and as the reference from [http://eigen.tuxfamily.org/dox-devel/group__TopicStructHavingEigenMembers.html](url)
this will be disposed automatically.

TSC21, 2018-10-31:
Ok cool. Can you issue a PR with that fix? Thanks!

GaoGeolone, 2018-10-31:
> 
> 
> Ok cool. Can you issue a PR with that fix? Thanks!

Excuse me, maybe I need some time to figure out how to use 'rosinstall_generator' to PR

GaoGeolone, 2018-10-31:
#1111 I made a PR, may it will help

GaoGeolone, 2018-10-31:
@TSC21 Excuse me ,I don't know why the Continuous Integration failed ,maybe it is the version issue?

closed 2018-11-06 by vooon