mavlink/mavros #1113: Distance Sensor rotation is incompatible with PX4

DanielePettenuzzo, 2018-11-08:
I'm having some issues in sending distance sensor data from PX4 to mavros. It seems that the rotations between PX4 and mavros are not consistent. When I receive distance_sensor messages via mavlink I get the following error in mavros:

```[ERROR] [1541510004.514463407]: DS: hrlv_ez4_pub: received sensor data has different orientation (PITCH_270) than in config (ROLL_180)!```

Pitch 270 is the orientation of the downwards facing distance sensor in PX4, while I think ROS expects Roll 180. 

How can I solve this?

DanielePettenuzzo, 2018-11-08:
FYI @TSC21 @baumanta @thomasgubler

mhkabir, 2018-11-08:
You have an old mavros config file which doesn't match what's on mavros master.

Update yours to match this : https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml

closed 2018-11-08 by mhkabir