mavlink/mavros #1126: MAVROS::Altitude still a valid message?

FaboNo, 2018-11-23:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am working on an altitude filter fusing imu, lidar and if possible altitude provided by the barometer.
From my understanding mavros::altitude is providing the altitude from the barometer.
I removed it from the blacklist and it appears when I do rostopic list:

fabrice@fabrice-N650DU:~$ rostopic list
/diagnostics
/mavlink/from
/mavlink/to
/mavros/adsb/send
/mavros/adsb/vehicle
/mavros/altitude
/mavros/battery
/mavros/cam_imu_sync/cam_imu_stamp
/mavros/distance_sensor/rangefinder_pub
/mavros/distance_sensor/rangefinder_sub
/mavros/extended_state
/mavros/fake_gps/mocap/tf
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/gp_lp_offset
/mavros/global_position/gp_origin
/mavros/global_position/home
/mavros/global_position/local
/mavros/global_position/raw/fix

But when I entered rostopic echo /mavros/altitude I got the following message:

<fabrice@fabrice-N650DU:~$ rostopic echo /mavros/altitude 
ERROR: Cannot load message class for [mavros_msgs/Altitude]. Are your messages built?>

This message is still valid, and if so how to built it?

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: 1.8.1

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```
roslaunch mavros apm2.launch 
WARNING: Catkin package name "droneAltitudeFiltering" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.
... logging to /home/fabrice/.ros/log/fbf17f58-ef2e-11e8-9797-b0359f2c094d/roslaunch-fabrice-N650DU-28226.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Catkin package name "droneAltitudeFiltering" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.
started roslaunch server http://fabrice-N650DU:46513/

SUMMARY
========

[... 192 lines cut ...]
[ INFO] [1542984572.065782396]: Plugin vision_speed_estimate blacklisted
[ INFO] [1542984572.065938960]: Plugin waypoint loaded
[ INFO] [1542984572.070658776]: Plugin waypoint initialized
[ INFO] [1542984572.070833061]: Plugin wind_estimation loaded
[ INFO] [1542984572.071577258]: Plugin wind_estimation initialized
[ INFO] [1542984572.071652263]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1542984572.071668395]: Built-in MAVLink package version: 2018.10.10
[ INFO] [1542984572.071688945]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1542984572.071703225]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1542984572.083378902]: IMU: High resolution IMU detected!
[ INFO] [1542984572.092522472]: IMU: Attitude quaternion IMU detected!
[ INFO] [1542984572.913443774]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1542984572.924178386]: IMU: High resolution IMU detected!
[ INFO] [1542984572.926135525]: IMU: Attitude quaternion IMU detected!
[ INFO] [1542984573.923436484]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1542984573.923506219]: VER: 1.1: Flight software:     010801ff (00000000FCA29321)
[ INFO] [1542984573.923535275]: VER: 1.1: Middleware software: 010801ff (00000000FCA29321)
[ INFO] [1542984573.923597331]: VER: 1.1: OS software:         071600ff (0000000000000000)
[ INFO] [1542984573.923638068]: VER: 1.1: Board hardware:      00000011
[ INFO] [1542984573.923670181]: VER: 1.1: VID/PID:             26ac:0011
[ INFO] [1542984573.923692816]: VER: 1.1: UID:                 3136510f34313630
[ WARN] [1542984573.932444279]: CMD: Unexpected command 520, result 0
[ INFO] [1542984582.914598468]: HP: requesting home position
[ INFO] [1542984587.920481393]: WP: mission received
[ INFO] [1542984592.914261692]: HP: requesting home position

trishantroy, 2018-12-06:
Are other topics working? Have you sourced the mavros package before running?

vooon, 2018-12-06:
Try to use fresh mavros, kinetic should have 0.27.

FaboNo, 2018-12-14:
@trishantroy @vooon thanks for your messages. Actually I got a fresh mavros as vooon suggested and it worked

closed 2018-12-14 by FaboNo