mavlink/mavros #1130: Question:  publishing pose setpoints in body reference frame [PX4 FCU]

francelico, 2018-12-04:
Hi,

I am a relatively new user of mavros and px4, and I am trying to publish positions setpoints by publishing into the setpoint_position/local topic. I got started using the optical flow iris gazebo model and the Python Offboard examples on dev.px4.io (using a slightly modified version of mavros_test_common to disable the GPS and home position checks).

I am trying to send setpoints relative to the current position of the drone. I thought of using subscribers to read the current local pose and publishing the setpoints relative to those but it seems very convoluted. I suspect changing the frame of reference to be body fixed should be simpler.

I am quite confused about the current coordinate frames used to publish the setpoints. At the moment positive x seems to correspond to North, positive y to West and positive z to Up, which does not correspond to any coordinate frames I ve encountered. In the example, the frame_id parameter is set to "base_footprint", but I have tested with any other string and it yields the same results.

What process should I follow to change the frame of reference to body fixed? Is there any example python code available that I can follow?

Maybe using setpoint_raw messages would work better, but I did not find a resource listing the bitmasks that should be used.

Thanks for your help.

sarahToonsi, 2018-12-28:
Hey if you found an answer for your question please share  :)

francelico, 2019-01-07:
Hi @sarahToonsi , in the end I went with publishing the setpoints in the ENU reference frame. I used the local position and orientation info and some trigonometry to convert relative setpoints into absolutes one in the ENU frame before publishing.

What is really confusing is that Gazebo uses a NWU reference frame, with 0 degrees pointing North, while MAVROS uses the ENU reference point, with 0 degrees pointing East. So make sure you convert when using Gazebo. In real flight ENU is automatically converted to NED so no need to worry about that.

Good luck!