mavlink/mavros #1131: "TM : RTT too high for timesync" error

trishantroy, 2018-12-05:
----
### Issue details
Hey.
I am using Pixhawk in my quadcopter. I am trying to connect Pixhawk to my laptop using a USB-TTL converter. I have connected the Pixhawk on the TELEM1 port. When I roslaunch mavros, I am getting an error about timesync. When I directly connected with USB, this error is not coming.
I have tried this configuration earlier and it was working them (the FW was APM though). 

Please help in resolving the issue. Thanks in advance.

### MAVROS version and platform

Mavros: How to check this? I installed mavros about a month ago 
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

PX4 1.8.1 [Default stable]

### Node logs

```
trishant@Eurus-Holmes:~/mavros_ws$ roslaunch mavros px4.launch 
... logging to /home/trishant/.ros/log/ab154086-f886-11e8-9f1a-2c6e8522ef61/roslaunch-Eurus-Holmes-31841.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Eurus-Holmes:37845/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
[... 379 lines cut ...]
      - 
        key: "Mean RTT (ms)"
        value: "16.234815"
      - 
        key: "Last remote time (s)"
        value: "348.116496000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---

```

### Check ID

```
trishant@Eurus-Holmes:~/mavros_ws$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 973 messages, from 1 addresses
sys:comp   list of messages
  1:1     147, 1, 36, 230, 0, 32, 105, 74, 44, 141, 77, 111, 241, 83, 340, 245, 22, 4, 30

```

mhkabir, 2018-12-05:
Not enough bandwidth on TELEM1 for timesync. Disable sys_time or switch to a higher baud link.

closed 2018-12-05 by mhkabir

trishantroy, 2018-12-05:
Thanks @mhkabir 
Using TELEM2 solved the problem.