mavlink/mavros #1132: No definition of [x] for OS [debian]

grcasanova, 2018-12-07:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
Mavros installation on rosbot image for Raspberry Pi 3 B, command 
`rosdep install --from-paths src --ignore-src -y` fails with

> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> image_geometry: No definition of [opencv3] for OS [debian]
> mavros: No definition of [geographic_msgs] for OS [debian]
> mavros_msgs: No definition of [geographic_msgs] for OS [debian]
> mavros_extras: No definition of [tf] for OS [debian]
> test_mavros: No definition of [control_toolbox] for OS [debian]
> cv_bridge: No definition of [opencv3] for OS [debian]

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

Version: 3.7.1

vooon, 2018-12-07:
That happens because rosdep repo don't have entries for `[x]`.
You can:
a) override OS distro
b) add path local rosdep yaml to /etc, then write file like that:
```yaml
#$ cat local_rosdep.yaml 
#libvtk:
#  ubuntu:
#    saucy: [libvtk5-dev]
#libflann:
#  ubuntu:
#    saucy: [libflann-dev]
#libglew-dev:
#  ubuntu:
#    saucy: [libglew-dev]
#gstreamer0.10-plugins-ugly:
#  ubuntu:
#    saucy: [gstreamer0.10-plugins-ugly]
#python-pyassimp:
#  ubuntu:
#    saucy: [python-pyassimp]
#libogre-dev:
#  ubuntu:
#    saucy: [libogre-1.8-dev]
gazebo:
  ubuntu: [gazebo5]
```

vooon, 2018-12-07:
Please note that this problem is not specific to mavros, it is general for ROS.

grcasanova, 2018-12-08:
Hi, thank you for your answer. I tried overriding the yaml file and OS (with all Debian and ubuntu distributions) yet I can't make it work.

I followed almost all guides from reputable sources on how to install ROS on a Raspberry Pi 3 but every time some unusual error pops out.

TSC21, 2018-12-08:
You have another solution which is installing the above dependencies by source and then ignore those dependencies on rosdep install.

master-25, 2018-12-28:
I actually used the rosinstall generator for installing the dependencies yet I cant get MAVROS to work on raspbian stretch
I have raspberry pi 3b and have installed ros kinetic mavlink is 2018.12.12