mavlink/mavros #1133: /mavros/imu/data linear_acceleration abnormal

kdzy, 2018-12-08:
### Issue details
I want to subscribe the topic /mavros/imu/data, but when I run "rostopic echo /mavros/imu/data", it appears as follows:
 
[image]
[image]

 The linear_acceleration is abnormal; it is either 0 or infinity. But when I use QGC to examine the value of acceleration, it is correct. I have tested multiple version of firmware, including 1.8.1, 1.7.0, 1.7.3, 1.6.5, 1.5.5, but they have the same results. Does anyone have the same problem? Is it the reason of mavros version? I will appreciate it if someone can help me.
[image]

### MAVROS version and platform

Mavros: 0.17.5
ROS: jade
Ubuntu: 14.04

### Autopilot type and version

[  ] ArduPilot
[X  ] PX4

Version: 1.7.0

rostopic echo  /diagnostics
[image]

vooon, 2018-12-08:
Try to build fresh version.

kdzy, 2018-12-10:
> Try to build fresh version.

Thanks for you reply.
I have tried to build another version of mavros. But because my processor is odroid, when I build another version, there are many errors. So I want to find whether there is another convenient way?

AlexWUrobot, 2023-02-05:
I have a similar simulation error
[image]