mavlink/mavros #1134: Not going in Position Hold mode when using Vision Position Estimate

trishantroy, 2018-12-08:
### Issue details
Hey,
I am using Pixhawk in my quadcopter. I have created a rosnode (/loc_vpe) to publish data on the /mavros/vision_pose/pose topic, taking x,y data from SLAM, z from TFMini (from /mavros/altitude topic) and orientation data from Pixhawk's imu (from /mavros/imu/data topic), time stamp: ros::Time::now(), header: "map" [Is this correct?]
When I run the node, I am getting data on the /mavros/local_position/pose topic.
I have set "EKF2_AID_MASK" as 8 ()tried with 24 as well) and "EKF2_HGT_MODE" as Range Sensor.

When I switch to Position mode using transmitter, there is a red light blinking instead of a green one. This means that the Pixhawk is unable to get it's position, right?
What should I do to solve the issue? Thanks in advance.

rqt_graph:
[image]

### MAVROS version and platform

Mavros: How to check this? Installed a month ago.
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] PX4: 1.8.1 px4fmu-v2 default

### Node logs

```
umic@umic-nuc-i5:~/mavros_ws$ roslaunch mavros px4.launch 
... logging to /home/umic/.ros/log/f5935d76-fae5-11e8-b36f-f8633f4a7ab3/roslaunch-umic-nuc-i5-7699.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.32:41163/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
[... 353 lines cut ...]
      - 
        key: "Mean dt (ms)"
        value: "0.016995"
      - 
        key: "Last system time (s)"
        value: "2004.282824000"
      - 
        key: "Time offset (s)"
        value: "1544271054.549984455"
---

```

### Check ID

```
umic@umic-nuc-i5:~$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 5185 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 132, 30, 31, 32, 36, 74, 77, 83, 340, 102, 230, 105, 111, 241, 245

```

trishantroy, 2018-12-08:
I am sorry. It is working now. Changing the frame_id to map solved the problem I think. It was base_link earlier.
Can the frame_id be the source of issue?

trishantroy, 2018-12-08:
I am sorry for opening the issue. It was working now! 
The only change I did was change the frame_id to map. It was base_link earlier. May this have been a possible reason for the problem earlier?

closed 2019-08-07 by trishantroy

vipin3690, 2022-10-17:
@trishantroy Where did you changed the frame id?
Can you please explain , i am having the same issue