mavlink/mavros #1137: Cannot Hover, drone pose provided by SLAM, Setpoint_position clarifications

FaboNo, 2018-12-14:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

In a first test, I wanted to take off and hover - the position of the drone is provided by a SLAM. 
I am using LPE, and Mavros to send the estimated position (see code below) and to send also the target position (see the code below).

For LPE:
`ATT_EXT_HDG_M` set to « vision »

 `LPE_FUSION`, I checked :
- Fuse vision positions
- Fuse vision yaw 
- Fuse Baro

Before doing real test, I launched all the processes, and check with the RC transmitter that Position mode is not rejected by PX4. I got the green light (if I can say that) so I went for a serie of real trials.

If you look at the pic below, you can see that the drone is take off and after circling around the the hovering position and then doing larger and larger circles. Despite the "circling" the SLAM is working quite well and provides a position which is correct but impossible for the drone to hover, it was unstable. So I would like to understand what is wrong here.

[image]

The possibilities I envisioned are:
- Failure of the SLAM : but from the results I got, it is working
- Tuning of the Attitude controler: I tried but it did not change anything
- Wrong coding or something I missed with mavros messages 

So, this why I am looking for clarifications regarding the Mavros messages.

On one end, I am sending the position computed by the SLAM with` /mavros/vision_pose/pose`, here is the code snippet:

```
                geometry_msgs::PoseStamped poseST_;
		poseST_.header.stamp = ros::Time::now();
		// poseST_.header.frame_id = 1;
                poseST_.header.seq = count;
		poseST_.pose.position.x  = pose.pose.position.x;
  		poseST_.pose.position.y  = pose.pose.position.y;
		poseST_.pose.position.z  = z_measured;
		poseST_.pose.orientation = tf::createQuaternionMsgFromRollPitchYaw(roll, pitch, yaw);
		//std::cout << "send pose to LPE \n";
		vision_pose_pub.publish(poseST_);
                count++;
```
Coordinates are in **ENU frame (or MAP frame which is similar)**  and thus z_measured is positive (Z Up).

At this stage I have two questions:
- Is it necessary to set `poseST_.header.frame_id` ? Indeed what is the difference between 0 (no frame) and (1) global frame, i.e, global_frame = map? Besides if I set frame_id = 0 I got an error from the compiler because frame_id is a string.
- Why do we need to increment a counter to set `poseST_.header.seq` , what is the purpose of this counter?

On the other side, there is a controler which is sending setpoint_position/local periodically, here is the code snippet:
```
while(ros::ok()){

		if( current_state.mode != "OFFBOARD" && (ros::Time::now() - last_request > ros::Duration(5.0))){
	 	 	if( set_mode_client.call(offb_set_mode) && offb_set_mode.response.mode_sent){
                		ROS_INFO("Offboard enabled");
            		}
            		last_request = ros::Time::now();
       		} else {
            		if( !current_state.armed && (ros::Time::now() - last_request > ros::Duration(5.0))){
               			if( arming_client.call(arm_cmd) && arm_cmd.response.success){
                    			ROS_INFO("Vehicle armed");
                		}
                		last_request = ros::Time::now();
            		}
        	}

[... 220 lines cut ...]
      - 
        key: Target frequency (Hz)
        value: 40.000000
      - 
        key: Minimum acceptable frequency (Hz)
        value: 38.000000
      - 
        key: Maximum acceptable frequency (Hz)
        value: 42.000000

```

### Check ID

```
rosrun mavros checkid

OK. I got messages from 1:1.

---
Received 6627 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 132, 140, 141, 147, 4, 26, 30, 31, 32, 33, 36, 74, 76, 83, 85, 102, 230, 105, 111, 241, 242, 245

```

YCH188, 2019-03-08:
@FaboNo Hello, I want to do the same thing to use slam for x, y coordinate. I want to know how do you use mavros to control UAV, use /mavros/setpoint_position/local? I publish the pose to /mavros/vision_pose/pose, but I don't konw how to control the UAV through mavros.

FaboNo, 2019-03-12:
@YCH188 Hi, you can use /mavros/setpoint_position/local if you want to control the drone in local position (with respect to the "map" frame) or you can also use mavros/setpoint_raw/local if you want to control the velocity (although in this case you will probably have to add you own PIDs) - both are working 
Regards

FaboNo, 2019-03-12:
I close the issue because the problem was related to the weigt of the drone .

closed 2019-03-12 by FaboNo