mavlink/mavros #1140: How to enable ExtendedState message?

Seanmatthews, 2018-12-19:
### Issue details

I see in the sys_status module that the ExtendedState message is handled. I know the sys_status module is on because I can see the typical state message. But I can't figure out why I'm not getting extended state. Are my versions too old? I note below that I'm running Indigo, and ExtendedState is a Kinetic addition, but I run Indigo with upstream versions of Mavros and Mavlink. Any ideas here?

### MAVROS version and platform

Mavros: 0.23.1
Mavlink: 2018.5.5-0
ROS: Indigo
Ubuntu: 14.04?

### Autopilot type and version

[  ] ArduPilot
[ x] PX4

Version: 1.8.0

TSC21, 2018-12-19:
Upstream version of MAVROS you mean 0.27?

TSC21, 2018-12-19:
Unless you build it from source, you won't have the upstream version on your Indigo system. We stopped supporting Indigo for a while now.

Seanmatthews, 2018-12-20:
So with my versions, I do see the ExtendedState message as capable of being published by sys_status, but I don't see the topic for it appear. Why doesn't the message get published?