mavlink/mavros #1141: TM: Clock skew detected problem

elgarbe, 2018-12-20:
### Issue details

I'm running px4 in SITL with gazebo together with another package for simulatin ASVs.
After a couple of minutes I get a message:
[WARN] [1545309362.033245977, 128.800000000]: TM: Clock skew detected (-9223372036.853929520 s). Hard syncing clocks.
and after that the timestamps of /uav1/mavros/global_position/local goes from a normal value (45, 70, whatever) to big number like 633438558
I'm using an intel i7 7700 with 8 cores, 8GB RAM and a GT710 graphic card. The 8 cores are loaded with 40 to 50%

### MAVROS version and platform

Mavros: 0.19.0
ROS: Kinetic
Xubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4 in SITL

Version: ?3.7.1?

### Node logs

```
[ INFO] [1545309343.317787886, 110.227000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309345.317189227, 112.212000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309347.316521099, 114.190000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309349.317020046, 116.176000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309351.317100802, 118.159000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309353.318881310, 120.143000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309355.320072555, 122.134000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309357.320017812, 124.121000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309359.319612287, 126.102000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309361.321273321, 128.092000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ WARN] [1545309362.033245977, 128.800000000]: TM: Clock skew detected (-9223372036.853929520 s). Hard syncing clocks.
[ INFO] [1545309363.322604505, 130.082000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309365.323691284, 132.066000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309367.325695954, 134.049000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309369.325697013, 136.027000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309371.326319817, 138.005000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309373.330800795, 139.988000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309375.330758064, 141.969000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309377.329715485, 143.946000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309379.332831832, 145.937000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ INFO] [1545309381.336239639, 147.924000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
[ WARN] [1545309382.522514058, 149.100000000]: TM: Clock skew detected (-0.011799176 s). Hard syncing clocks.
WARN  [mavlink] [timesync] Hard setting offset.
[ INFO] [1545309383.349594284, 149.914000000]: HP: Home lat 47.398639, long 8.548792, alt 489.126007
```

### Diagnostics

```
elgarbe@i7:~$ rostopic echo -n1 /diagnostics
header: 
  seq: 169
  stamp: 
    secs: 171
    nsecs: 500000000
  frame_id: ''
status: 
  - 
    level: 0
    name: "uav1/mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Received packets:"
        value: "22413"
[... 310 lines cut ...]
      - 
        key: "Last system time (s)"
        value: "210.529619000"
      - 
        key: "Time offset (s)"
        value: "9223372036.225246429"
---
elgarbe@i7:~$ 

```

### Check ID

```
elgarbe@i7:~$ rosrun mavros checkid
WARNING: /mavros/target_system_id not set. Used default value: 1
WARNING: /mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

```

vooon, 2018-12-20:
Video is unavailable. CPU load looks too high, but do you sure that load is from mavros? PX4 SITL may generate lots of load.
Usually clock skew happen when you have too high RTT. But that shouldn't be the case for sitl.
Also version 0.19 is pretty old, i recommend to update.

mhkabir, 2018-12-20:
The newer version should have a better timesync algorithm.

elgarbe, 2018-12-20:
video is working now
https://www.youtube.com/watch?v=JeY1U5V7YJY

The first launch is for an USV simulator it is the source of CPU load.

I will try with a newer version of mavros. Thank!

elgarbe, 2018-12-20:
It is working much much better with lastest version.
It seems that some times mavros detect a time jump and automatically reset de sinchronizer (there are some messages like that).
Loading a litle more my PC make the timestamp to jump to 633xxxxx but after a few seconds it resets the stamps to a correct value. But after that mavros timestamps lags 3 or 4 seconds messages from other packages.
I will work to reduce the load of the simulator.

Thank

mhkabir, 2018-12-20:
This is also possibly because the sync algorithm is not tuned for a system which is actively correcting its reference clock using NTP. 

Can you try to disable It? For Ubuntu : `sudo timedatectl set-ntp no`

elgarbe, 2018-12-20:
I run that comand and relaunch my simulation.
I made a mistake mavros timestamps are a litle faster that other packages. I see sec:170 for my package and sec: 177 for mavros messages... It could be a new problem, because I have a package that need to receive position messages from UAV (mavros) and position messages for a USV from another package and they need to be time synchronized. 
After 200 seconds mavros timestamp jump to 6xxxxxxx sec and this time it is not reseted to correct value....  :-(

elgarbe, 2019-04-03:
Hi, after a couple of several test the problem is not completely solved.
Sooner or later, mavros timestamp jumps to a much bigger number.
Is there a way to get a more stable timestamp? what is the source of the problem here? why the timestamp jumps?

Thank!

spurnvoj, 2026-10-09:
Still present in 2.14.0 and current `ros2`. I think I found the cause.

In PX4 SITL with `use_sim_time`, ROS and PX4 share a clock, so the real offset is ~0 and often slightly negative. `handle_timesync()` computes it as `(ts1 + now_ns - tc1 * 2) / 2` with `now_ns` unsigned, so a negative offset wraps to ~2^63 ns (the `-9223372036.85 s` above). The filter occasionally converges on that and publishes it, and then `synchronise_stamp()` gives huge or negative stamps.

Real hardware isn't affected: there the offset is large and positive.

Fix in #2316: keep the offset signed (`int64_t`). With it, the offset in SITL stays within a few ms of 0 and the filter never resets.

closed 2026-10-21 by vooon