mavlink/mavros #1149: Publishing mavros/rc/in topic to mavros_msgs/RCIn

TK1109, 2019-01-02:
Hello,

I am trying to get mavros/rc/in to be published as a message: mavros_msg/RCIn
I have entered the following into a terminal window: 
$rostopic echo /mavros/rc/in | rostopic pub bar mavros_msgs/RCIn

Ultimately I want to be able to read the channel values as they change with input in MATLAB.

However, when I look in MATLAB to see if the information can be read or not, it appears that no new information is read by MATLAB from the mavros_msgs/RCIn

I am I entering the publish command line correctly? Is there a way to verify in ROS that the message is publishing? I am quite new to ROS and I am having trouble navigating this issue.

Thanks!

[image]

vooon, 2019-01-03:
I don't know matlab, but perhaps you mess with topic type (`mavros/RCIn`) and topic name?
Matlab code looks like you create new empty RCIn massage (struct/class) and print it (new message are zeroed).

And why you need to republish topic? Maybe easier just remap it?

vooon, 2019-01-03:
Oh, also you can use RQT to watch what topics exists and what data is published.

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248