mavlink/mavros #1153: Unable to arm APM with with ROS

Alvinosaur, 2019-01-04:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I have a couple of questions regarding the proper way to connect to my pixhawk (running on APM) correctly and also how to arm and send commands through an offboard computer. I am also unable to arm the pixhawk through mavros. 

First, I am only able to occasionally connect with the pixhawk with mavros and get actual data on topics. My procedure for connecting is as follows:
1. Connect battery to drone and usb serial to my laptop (just temporary for testing)
2. run `roscore` in one tab
3. run mavros: `roslaunch mavros apm.launch`

After the above steps, I feel that I should be able to connect, but the various topics I can use (ex: `/mavros/battery`) do not output anything. When I see this, I try random things like using the RC to try to connect and arm the drone or starting up QGC, which successfully connects with the drone. After several of these random attempts and head-banging, I can connect. 

Second, I am unable to arm the drone. I try running the following: `rosrun mavros mavsafety arm`, but this produces the error: `Request failed. Check mavros logs`. Of course then, I try to check the logs by first running `roslaunch-logs`, which gives me the path to a file that lists various log files. I vim into these, but do not see anything meaningful. I then run `rqt_console` to see real-time error messages, which is pretty helpful, but doesn't help me with arming the drone. Where can I check the mavros logs? 

Also, when I am trying to arm the drone through a .cpp file rather than through terminal, I am following the [offboard example](https://dev.px4.io/en/ros/mavros_offboard.html). Here is my code below. Before taking a look, my understanding is that I publish desired values (ex: global positions) that I want the drone to move to. I also should publish some arm command to the arming topic, `mavros/cmd/arming`. 

Please, any assistance is greatly appreciated. Please let me know if my post was bad as this is my first github issue post. This project is to have an Iris 3DR drone fly to some goal location and detect crop anomalies to send to mobile ground robots for further inspection :) Thank you!

```
#include <ros/ros.h>
#include <geometry_msgs/PoseStamped.h>
#include <geometry_msgs/TwistStamped.h>
#include <geometry_msgs/Vector3.h>
#include <mavros_msgs/CommandBool.h>
#include <mavros_msgs/SetMode.h>
#include <mavros_msgs/State.h>
#include <mavros_msgs/Thrust.h>

#include <sstream>

mavros_msgs::State current_state;
void state_cb(const mavros_msgs::State::ConstPtr& msg){
    current_state = *msg;
}

int main(int argc, char **argv)
{
    char *target_mode = "GUIDED";
    ros::init(argc, argv, "offb_node");
    ros::NodeHandle nh;

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
            ("mavros/state", 10, state_cb);
    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_position/local", 10);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
            ("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
            ("mavros/set_mode");

    ros::Publisher pub_att = nh.advertise<geometry_msgs::PoseStamped>("/mavros/setpoint_attitude/attitude", 100);
    ros::Publisher pub_vel = nh.advertise<geometry_msgs::TwistStamped>("/mavros/setpoint_attitude/cmd_vel", 100);
    ros::Publisher pub_thr = nh.advertise<mavros_msgs::Thrust>("/mavros/setpoint_attitude/thrust", 100);

    //the setpoint publishing rate MUST be faster than 2Hz
    ros::Rate rate(20.0);

    // wait for FCU connection
    while(ros::ok() && !current_state.connected){
        ros::spinOnce();
        rate.sleep();
        // std::stringstream ss;
        // ss << "[Not Connected]" << ", armed: " << armed << mode;
        ROS_INFO("Not Connected!");
    }

[... 263 lines cut ...]
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Voltage"
        value: "11.74"
      - 
        key: "Current"
        value: "0.3"
      - 
        key: "Remaining"
        value: "91.0"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 878 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 133, 136, 152, 150, 24, 27, 29, 30, 33, 162, 163, 36, 165, 42, 178, 62, 74, 35, 116, 125
```

Alvinosaur, 2019-01-10:
I've solved this issue, but am having others. I apologize for opening the issue so soon.

closed 2019-01-10 by Alvinosaur

Fisherman3, 2019-10-31:
Sorry to reopen a topic that has been closed for so long, but how did you solve the issue? I'm having similar problems.

Jawad-RoboLearn, 2021-12-30:
Hi, how did you solve this issue? I have faced the same problem. As I have used the same type of code. 
1. First I have used vicon for external pose. In launch file, i remap from /vicon/myquad/myquad to /mavros/mocap/tf. It works well. 
2. Then, i use /mavros/setpoint_position/local to send the message. 
3. The issue is offboard is enable after every 5 seconds but no mavros did not arm the drone. 
4. Whats wrong?