mavlink/mavros #1154: Cannot control actuators individually

JohannLange, 2019-01-04:
### Issue details

Hello everyone,
I want to control each actuator of my quadrotor individually. The goal is, to send a signal to each actuator between 0 and +1 to control the speed. If my approch is totally wrong, or there already exists another way, please point me in the right direction.

To achieve my goal I wrote a mixer file (see end of issue details). Now, whenever I load my mixer in the NSH shell in QGC via ```mixer load /dev/pwm_output0 /etc/mixers/my_mixer.main.mix```, three of my four actuators start to spin, but not the fourth one. The same behaviour happens, when I'm using the ```IO_pass.main.mix```.
Moreover, I cannot make them stop spinning via MAVROS. To try that, I used the follwoing function/message using the ```/mavros/target_actuator_control```-Topic (see the end again).
Using my remote, I am able to change the speed of the actuators (arming also activates the fourth actuator).

Can anyone help me, controlling the actuators individually via MAVROS?
Any help is deeply appreciated.

#### Mixer
```
Servo 0
-------------
M: 1
S: 0 0  10000  10000      0 -10000  10000

Servo 1
-------------
M: 1
S: 0 1  10000  10000      0 -10000  10000

Servo 2
-------------
M: 1
S: 0 2  10000  10000      0 -10000  10000

Servo 3
-------------
M: 1
S: 0 3  10000  10000      0 -10000  10000
```

#### MAVROS Message definition
```
void send_Actuator_Control_Handler(ros::NodeHandle node, double a0 = 0.0, double a1 = 0.0, double a2 = 0.0, double a3 = 0.0, double a4 = 0.0, double a5 = 0.0, double a6 = 0.0, double a7 = 0.0){
    pub_act_control = node.advertise<mavros_msgs::ActuatorControl>("/mavros/target_actuator_control", omh_loop_rate_);

    mavros_msgs::ActuatorControl act_control_msg;

    act_control_msg.header.stamp = ros::Time::now();
    act_control_msg.header.seq = act_msg_count;
    act_control_msg.header.frame_id = "1";
    act_control_msg.group_mix = 1;
    act_control_msg.controls[0] = a0;
    act_control_msg.controls[1] = a1;
    act_control_msg.controls[2] = a2;
    act_control_msg.controls[3] = a3;
    act_control_msg.controls[4] = a4;
    act_control_msg.controls[5] = a5;
    act_control_msg.controls[6] = a6;
    act_control_msg.controls[7] = a7;

    pub_act_control.publish(act_control_msg);

    ++act_msg_count;
}
```

### MAVROS version and platform

Mavros: 0.25.1
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[... 389 lines cut ...]
      -
        key: "Last RTT (ms)"
        value: "24.049965"
      -
        key: "Mean RTT (ms)"
        value: "75.818239"
      -
        key: "Last remote time (s)"
        value: "1611.755541000"
      -
        key: "Estimated time offset (s)"
        value: "0.000000000"
---
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 1002 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 36, 230, 147, 32, 105, 74, 141, 111, 241, 83, 340, 245, 22, 4, 30, 31
```

robin-ecn, 2019-04-04:
Do you have any news?

alexisutc, 2019-04-09:
up

alexisutc, 2019-05-24:
Hi,

I solved the problem by writing a costum node that uses directly the PWM of my microcontroller (Navio2 in my case)
also you need to disable the writing of the pwm in the ardupilot, else there is an interference.

ogKamy, 2019-07-08:
How did you disable the writing of the pwn in ardupilot?

alexisutc, 2019-07-09:
You have to change the code of the ardupilot:

SVR_Channel.h : change value of servochannel (for me from 16 to 8)
SVR_Channels.cpp : comment corresponding group info part corresponding to unused servo channels

-> compile
-> transfer

JohannLange, 2020-12-18:
While I never came around solving the issue, the firmware I used is now deprecated and a lot of things changed.
As my question is, in it's current form, no longer relevant, I'll close the issue.

closed 2020-12-18 by JohannLange