mavlink/mavros #1155: "rostopic echo /mavlink/to"

kevinjhur, 2019-01-07:
Currently I've tried to get the MAVLINK messages from GCS to FCU via mavros.
When I run "rostopic echo /mavlink/from" to get msg sending from FCU, it works fine.
But on running "rostopic echo /mavlink/to" to get msg send from GCS, it dosn't work.

Also when I run below test code, mavlink_from_cb() is called correctly by sending msg from FCU.
But mavlink_to_cb() is not called.
Is it normal process? If not, what is wrong for me? 

I want monitor mavlink message which is sent by GCS via mavros, is it possible?

void FCUIF::mavlink_from_cb(const mavros_msgs::Mavlink& msg)
{
    mavlink::mavlink_message_t mavMsg;
    mavros_msgs::mavlink::convert(msg, mavMsg);

    if(mavMsg.sysid == 1)
    {
        ROS_INFO("[MAVLINK MSG to FC] magic=%2X len=%2X inc=%2X comp=%2X seq=%2X sysid=%2X compid=%2X msgid=%6X chksum=%4d",
                 mavMsg.magic, mavMsg.len, mavMsg.incompat_flags, mavMsg.compat_flags, mavMsg.seq, mavMsg.sysid, mavMsg.compid, mavMsg.msgid, mavMsg.checksum);
    }
}

void FCUIF::mavlink_to_cb(const mavros_msgs::Mavlink& msg)
{
    mavlink::mavlink_message_t mavMsg;
    mavros_msgs::mavlink::convert(msg, mavMsg);

    if(mavMsg.sysid != 1)
    {
        ROS_INFO("[MAVLINK MSG to FC] magic=%2X len=%2X inc=%2X comp=%2X seq=%2X sysid=%2X compid=%2X msgid=%6X chksum=%4d",
                 mavMsg.magic, mavMsg.len, mavMsg.incompat_flags, mavMsg.compat_flags, mavMsg.seq, mavMsg.sysid, mavMsg.compid, mavMsg.msgid, mavMsg.checksum);
    }
}

bool FCUIF::initDrone(ros::NodeHandle nh)
{
    //the setpoint publishing rate MUST be faster than 2Hz
    ros::Rate rate(20.0);

    mavlink_from_sub      = nh.subscribe                                ("mavlink/from", 1000, &APMIF::mavlink_from_cb, this); // MSG read from FCU
    mavlink_to_sub        = nh.subscribe                                ("mavlink/to", 1000, &APMIF::mavlink_to_cb, this); // MSG read from GCS

	// wait for FCU connection
	while (ros::ok() && current_state.connected) {
		ros::spinOnce();
		rate.sleep();
	}

    return true;
}

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[X] ArduPilot
[X] PX4

Version: ?any?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo /mavlink/to
```

### Check ID

```
rosrun mavros checkid
```

TSC21, 2019-01-07:
Check https://github.com/mavlink/mavros/pull/1117.

closed 2019-01-07 by TSC21

kevinjhur, 2019-01-07:
> Check #1117.

I can't understand the relation between #1117 and my issue(#1155).
Please let me know why you close this post.