mavlink/mavros #1158: Setpoint_Position/Local Position Control: Relative Positioning

SchaeferB1, 2019-01-22:
----

### Issue details

Hello Everyone,
I am attempting to control a quadcopter‘s position using a camera to move to a target using the difference between the current location of the drone and the centroid of a target. To do this, I am using the current position from `~local_position/pose`, and subtracting a value from the current position that would then be published to `~setpoint_position/local`.  
The problem is the position value from `~local_position/pose` resets to approximately (0,0,0) in (x,y,z) when running the script every time, making the `~local_position/pose` values reset. After running the script once, the position from `~local_position/pose` seems to no longer be in the same frame. When publishing to the `~setpoint_position/local` topic, coordinates are published in a coordinate frame with an origin centered at the location of where the flight controller was booted. Although both topics use the ENU coordinate frame, the origin of the `~local_position/pose` position is changed every time the script is run. Is there any possible way to get local position data that does not reset every time a new script is run? 
The posted script replaces the camera data that would be used in lines 146-147 with fixed constant values for testing. When the script is run, the drone moves to (5m,5m,3m) and lands. When the script is run again, it takes off and remains in the same (x,y) position. Other posts suggest changing the apm_config.yaml. Does changing the frame of `~setpoint_position/local` in this configuration file change the frame that the input values are converted to?

### MAVROS version and platform

Mavros: ?0.18.4?
Version of Mavros was built into an image for the Navio2 Flight Controller. I believe this is the version of Mavros we are using: https://docs.emlid.com/navio2/common/ardupilot/configuring-raspberry-pi/. 
ROS: Kinetic
Raspbian: GNU/Linux 9

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: 3.5

### Node logs

```
rosrun mavros mavros_node _fcu_url:=udp://:14650@ _gcs_url:=udp://:14551@192.168.1.90:14550
[ INFO] [1548171941.196073151]: FCU URL: udp://:14650@
[ INFO] [1548171941.207366747]: udp0: Bind address: 0.0.0.0:14650
[ INFO] [1548171941.208232051]: GCS URL: udp://:14551@192.168.1.90:14550
[ INFO] [1548171941.208940691]: udp1: Bind address: 0.0.0.0:14551
[ INFO] [1548171941.209511987]: udp1: Remote address: 192.168.1.90:14550
[ INFO] [1548171941.380572463]: Plugin 3dr_radio loaded
[ INFO] [1548171941.389768786]: Plugin 3dr_radio initialized
[ INFO] [1548171941.390560185]: Plugin actuator_control loaded
[ INFO] [1548171941.417157026]: Plugin actuator_control initialized
[ INFO] [1548171941.418110559]: Plugin altitude loaded
[ INFO] [1548171941.426639127]: Plugin altitude initialized
[ INFO] [1548171941.427587192]: Plugin command loaded
[ INFO] [1548171941.464382795]: Plugin command initialized
[ INFO] [1548171941.465236745]: Plugin ftp loaded
[ INFO] [1548171941.498379360]: udp0: Remote address: 127.0.0.1:48247
[ INFO] [1548171941.522233466]: Plugin ftp initialized
[ INFO] [1548171941.523215749]: Plugin global_position loaded
[ INFO] [1548171941.635482280]: Plugin global_position initialized
[ INFO] [1548171941.636533677]: Plugin hil loaded
[ INFO] [1548171941.755061662]: Plugin hil initialized
[ INFO] [1548171941.756035038]: Plugin home_position loaded
[ INFO] [1548171941.788141152]: Plugin home_position initialized
[ INFO] [1548171941.789200049]: Plugin imu loaded
[ INFO] [1548171941.840590415]: Plugin imu initialized
[ INFO] [1548171941.841528115]: Plugin local_position loaded
[ INFO] [1548171941.873298034]: Plugin local_position initialized
[ INFO] [1548171941.874234224]: Plugin manual_control loaded
[ INFO] [1548171941.901421997]: Plugin manual_control initialized
[ INFO] [1548171941.902161522]: Plugin param loaded
[ INFO] [1548171941.922325609]: Plugin param initialized
[ INFO] [1548171941.923155081]: Plugin rc_io loaded
[ INFO] [1548171941.954899636]: Plugin rc_io initialized
[ INFO] [1548171941.955912648]: Plugin safety_area loaded
[ INFO] [1548171941.991687739]: Plugin safety_area initialized
[ INFO] [1548171941.992652730]: Plugin setpoint_accel loaded
[ INFO] [1548171942.019251343]: Plugin setpoint_accel initialized
[ INFO] [1548171942.020367583]: Plugin setpoint_attitude loaded
[ INFO] [1548171942.089494976]: Plugin setpoint_attitude initialized
[ INFO] [1548171942.090385020]: Plugin setpoint_position loaded
[ INFO] [1548171942.200324127]: Plugin setpoint_position initialized
[ INFO] [1548171942.201273858]: Plugin setpoint_raw loaded
[ INFO] [1548171942.281097821]: Plugin setpoint_raw initialized
[ INFO] [1548171942.282215935]: Plugin setpoint_velocity loaded
[... 162 lines cut ...]
    hardware_id: "udp://:14650@"
    values:
      -
        key: "Voltage"
        value: "-1.00"
      -
        key: "Current"
        value: "0.0"
      -
        key: "Remaining"
        value: "0.0"
```

### Check ID

```
WARNING: mavros/target_system_id not set. Used default value: 1
WARNING: mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
OK. I got messages from 1:1.

---
Received 16 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 77```

SchaeferB1, 2019-01-24:
@vooon @TSC21 I'd really appreciate your expertise in this matter, if possible. Thank you!

YCH188, 2019-03-08:
@SchaeferB1 From what I know you should publish the camera position to /mavros/vision_pose/pose, can you leave your email for further communication. I want to do something similar.

TSC21, 2019-03-08:
@YCH188 stop spamming the issues list please. Create your own issue of you want to ask qiestions, but only after you empty the other options, which is asking your questions on the appropriate forums.

YCH188, 2019-03-09:
@TSC21 Sorry for bothering, I will start a new issue. I didin't find the forums for mavros and the tutorials only contains simulation which makes me puzzled for real flight.

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248

erikasnow, 2020-02-16:
@SchaeferB1 I'm trying to do something very similar to what you described. However, I can't seem to get the setpoint_position/local message through to the pixhawk. The topic updates correctly, but I'm not seeing the corresponding message in Mavlink Inspector. Did you ever run into any problems like this?

ericjohnson97, 2020-02-16:
There are a couple things that need to happen when using setpoint position local. some of the things that I struggled with when learning mavros were:
- for ardupilot the drone must first be in the air to use setpoint position local 
- on an older version of mavros the origin for the setpoint position local plugin origin was set to the center of the earth instead of the initial GPS pose. this should have been fixed in later releases after june 2019

Here are some good tutorials using mavros for local navigation with some example code.
https://github.com/Intelligent-Quads/iq_tutorials

these are also some good reads 
https://github.com/mavlink/mavros/issues/906

https://github.com/mavlink/mavros/issues/1213