mavlink/mavros #1162: Redundant frames local_origin_ned and fcu_frd

okalachev, 2019-01-26:
`uas_data.cpp` module publishes some redundant static transformations: `local_origin` -> `local_origin_ned`, `fcu` -> `fcu_frd`, see https://github.com/mavlink/mavros/blob/master/mavros/src/lib/uas_data.cpp#L55.

First, with the default settings they are not even connected to anything, because there are no `local_origin` and `fcu` frames (they used to, but they were renamed).

Second, they are not standard for ROS conventions. If someone needs it, they can create static transforms themselves with `tf2_ros/static_transform_publisher`.

I propose to remove them at all, or (if the backwards compatibility matters here) at least to make some parameter to disable them.

TSC21, 2019-01-26:
@okalachev noted and I was already planning fix that. They will not be removed but rather be created only when odometey data is received from the FCU and handled on the odom plugin. I was add a PR when able.

TSC21, 2019-01-26:
Also that tf is only required on the odom plugin currently. So it can't be removed but it can rather be only published on the odometry handler.

okalachev, 2019-01-26:
So they will be removed from uas_data?

But why to add them to odometry plugin? Backwards compatibility? I don't think they are needed otherwise, they would only confuse users.

TSC21, 2019-01-26:
You need them. Try to use the odometry and you will understand that you need the opposite transform while applying the frame transforms there. The odom plugin does not use programatically set static transforms as the rest of the codebase of MAVROS.

TSC21, 2019-01-26:
The only transform being removed from uas_data is the fcu_frd->fcu, at least for now. There is no confusion to the users because all of these static transforms are only used on the odom plugin. They are not used anywhere else. The idea of putting it on uas_data was to have a throughout support for tf lookups when doing frame transforms. But that idea won't become a reality anytime soon. So it will be even better to just move all of the static transforms to where they are used. That will for sure remove any confusion.

okalachev, 2019-02-02:
>There is no confusion to the users because all of these static transforms are only used on the odom plugin.

But they are exposed as TF static transforms, so they are visible from rviz, rqt_tf_tree and other tools.

ifaceyou123, 2019-02-21:
Hi Guys,  when I run the checkout vision of v1.8.2 with this command roslaunch px4 mavros_posix_sitl.launch. I found a very strange problem. The tf tree contains a loop, Frame fcu_frd exists with parent fcu and frame fcu also exists with parent fcu_frd. How is this going? I got always a warning that the TF tree is invalid. Do I need to take care of this warning?
[image]

TSC21, 2019-02-21:
@ifaceyou123 this has been discussed on this issue. It's a warning yes, but bot nothing to worry. Should be fixed asap.

ifaceyou123, 2019-02-21:
@TSC21 Thanks a lot