mavlink/mavros #1168: setpoint_position/local: No Error, quad Not Responding

yuchnw, 2019-01-30:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I was trying to set target position to the iris quad in Gazebo, so I publish the x, y, z value to the /mavros/setpoint_position/local topic. The topic did receive these values (updated in echo command), however, the quad didn't move at all. Does anyone have idea what might cause this issue? I am new to PX4 so I guess I might missed some parts. Thanks!

### MAVROS version and platform

Mavros: latest released version
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

```
def setLocation():
       goalPos = PoseStamped()
       goalPos.pose.position.x = 0
       goalPos.pose.position.y = 0
       goalPos.pose.position.z = 15
       goalPos.header.stamp = rospy.Time.now()
       goalPos.header.frame_id = 'map'

       locPub.publish(goalPos)

def globalPositionCallback(globalPos):
    global latitude
    global longitude
    global altitude
    latitude = globalPos.latitude
    altitude = globalPos.altitude
    longitude = globalPos.longitude

def localPositionCallback(localPos):
    global x
    global y
    global z
    x = localPos.pose.position.x
    y = localPos.pose.position.y
    z = localPos.pose.position.z
    
def userInput():
    enter ='1'
    while ((not rospy.is_shutdown())and (enter in ['1','2','3','4','x'])):
        print "1: Set mode to ARM"
        print "2: Set mode to DISARM"
        print "3: Set mode to TAKEOFF"
        print "4: Set mode to LAND"
        print "x: Print GPS info"
        enter = raw_input("Enter your input: ")
        if (enter=='1'):
            ArmIt()
        elif(enter=='2'):
            DisarmIt()
        elif(enter=='3'):
            setLocation()
        elif(enter=='4'):
            LandIt()
        elif(enter=='x'):
            global latitude
            global longitude
            global altitude
            global x
            global y
            global z
            print ("longitude: %.7f" %longitude)
            print ("latitude: %.7f" %latitude)
            print ("altitude: %.7f" %altitude)
            print ("x: %.5f" %x)
            print ("y: %.5f" %y)
            print ("z: %.5f" %z)
        elif(enter=='c'): 
            break
        else:
            print "Invalid command!"

if __name__ == '__main__':
    rospy.init_node('iris_node', anonymous=True)
    rospy.Subscriber("/mavros/global_position/raw/fix", NavSatFix, globalPositionCallback)
    rospy.Subscriber("/mavros/local_position/pose", PoseStamped, localPositionCallback)
    locPub = rospy.Publisher('/mavros/setpoint_position/local',PoseStamped,queue_size=10) 

    userInput()
```

### rostopic echo /mavros/setpoint_position/local
```
header: 
  seq: 2
  stamp: 
    secs: 3834
    nsecs: 264000000
  frame_id: "map"
pose: 
  position: 
    x: 0.0
    y: 0.0
    z: 15.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0

```

### Diagnostics

I ran /diagnostics and all entries were normal.

### Check ID
Nothing shows
```
rosrun mavros checkid
```
@vooon @TSC21 Thanks!

Jaeyoung-Lim, 2019-01-30:
@yuchnw Were you able to switch to offboard mode?

yuchnw, 2019-01-30:
@Jaeyoung-Lim Yes I just tried it and I can switch to offboard mode as shown below:
```
rosrun mavros mavsetp local -v 0 0 0 0
Requesting OFFBOARD mode...
OFFBOARD mode are set.
```
Seems like I have connection to the flight controller since I can send command to make it arm/disarm/land. The only command that doesn't work is /setpoint_position/local.

yuchnw, 2019-01-30:
@Jaeyoung-Lim And another part I noticed might be weird is that when I ran `roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"` first and then `roslaunch px4 mavros_posix_sitl.launch`, the first one would die and give errors saying:
```
[ WARN] [1548863792.801206282]: Shutdown request received.
[ WARN] [1548863792.801262605]: Reason given for shutdown: [new node registered with same name]
============================================================================
REQUIRED process [mavros-1] has died!
process has finished cleanly
log file: /home/yuchen/.ros/log/a3772e1a-24a6-11e9-a071-7470fdecbdf5/mavros-1*.log
Initiating shutdown!
==========================================================================
```
I am not sure if I did anything wrong but these errors just make me confused.

Thanks!

TSC21, 2019-01-30:
Where are you actually using `setLocation()`?

TSC21, 2019-01-30:
> @TSC21 I called this function in my `userInput()` loop. So once I type '3' in the prompt, it will call `setLocation()`.

I don't think the logic above is right. By what you show above, you will have to keep pressing `3` so to keep the value of `enter` equals to `3`, right? Is that what you intend to? Because for sending setpoints, you need to have a constant stream at least at 2Hz. I advise you to read https://docs.px4.io/en/flight_modes/offboard.html and take a look at https://github.com/mavlink/mavros/blob/master/test_mavros/scripts/setpoint_position_demo.

closed 2019-01-30 by TSC21

yuchnw, 2019-01-30:
@TSC21 I have tried to just run the `setpoint_position_demo` however it also didn't work. Here is the screenshot of my terminals when running the demo script.

[image]

Looks like the `setpoint_position/local` topic received the parameters but the quad's not moving.

TSC21, 2019-01-30:
The example I gave you is not tested in a while but since it's a python script it could give you an idea of what you are doing wrong. The logic on your code, as I said, doesn't seem right

yuchnw, 2019-02-10:
@TSC21 @Jaeyoung-Lim Hi. I opened up the DOT graph of my node when I tried to publish to `/mavros/setpoint_position/local` and it seems like the connection has been established and continuous. However, the quad stays still. I tried run it after it takes off but got no luck. 
[image]
Do you have any idea what might cause this issue? I would appreciate your help!
Thanks!