mavlink/mavros #1172: ROS_INFO: called mavros trigger service! Success: 0, Result? 3

gladys0313, 2019-01-31:
Hi all,

I'm using [pixracer autopilot](https://docs.px4.io/en/flight_controller/pixracer.html) and a [ptgrey camera ](https://www.ptgrey.com/blackfly-s-mono-04-mp-usb3-vision-sony-imx287-2)(spinnaker camera driver) to run a VIO algorithm [ROVIO](https://github.com/ethz-asl/rovio). When running the launch file, there is in info like this:
`[INFO] called mavros trigger service! Success? 0 Result? 3`
The code regarding this info from spinnaker camera driver is like this:
`void startMavrosTriggering() {
    // First subscribe to the messages so we don't miss any.'
    sequence_time_map_.clear();
    trigger_sequence_offset_ = 0;

    if (!multi_camera_mode_) {
      const std::string mavros_trigger_service = "mavros/cmd/trigger_control";
      if (ros::service::exists(mavros_trigger_service, false)) {
        mavros_msgs::CommandTriggerControl req;
        req.request.trigger_enable = true;
        // This is NOT integration time, this is actually the sequence reset.
        req.request.cycle_time = 1.0;

        ros::service::call(mavros_trigger_service, req);

        ROS_INFO("Called mavros trigger service! Success? %d Result? %d",
                 req.response.success, req.response.result);
      } else {
        ROS_WARN("Mavros service not available!");
      }
    }`
Obviously the mavros trigger service is not called successfully, but I have no idea about the result code `3`. Does anyone know what "3" refers to? Thanks a lot!

vooon, 2019-02-01:
As i remember: https://mavlink.io/en/messages/common.html#MAV_RESULT

gladys0313, 2019-02-02:
@vooon it looks really promising! thanks. I guess it is "MAV_RESULT" but not "MAV_MISSION_RESULT"?

closed 2019-02-03 by TSC21