mavlink/mavros #1180: offboard request failed

alexisutc, 2019-02-15:
Hello world,

i'm new with ros and mavros. This is my config :
- My system uses mavros p4x (i'm not sure about it actually)
- I use a navio2
Actually i have following problem :
- I want to uses the actuator_control node to crontrol directly the PWM of my navio2, but actually i cant. I read that I have to go in the offboard mode... but how ? 
I tried
 rosrun mavros mavsys mode -c OFFBOARD
But it tells me 
Request failed. Check mavros logs

so i put here my log.
Can you please help me on that ?

thancks beforehand.
----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Kinetic
Rasbian

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
header:
  seq: 2710
  stamp:
    secs: 1550241520
    nsecs: 980842850
  frame_id: ''
status:
  -
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14650@"
    values:
      -
        key: "Received packets:"
        value: "10445"
      -
        key: "Dropped packets:"
        value: "0"
      -
        key: "Buffer overruns:"
        value: "0"
      -
        key: "Parse errors:"
        value: "0"
      -
        key: "Rx sequence number:"
        value: "62"
      -
        key: "Tx sequence number:"
        value: "83"
      -
        key: "Rx total bytes:"
        value: "4536058"
      -
        key: "Tx total bytes:"
        value: "661076"
      -
        key: "Rx speed:"
        value: "3601.000000"
      -
        key: "Tx speed:"
        value: "474.000000"
  -
    level: 0
    name: "mavros: GPS"
    message: "3D fix"
    hardware_id: "udp://:14650@"
    values:
      -
        key: "Satellites visible"
        value: "9"
      -
        key: "Fix type"
        value: "3"
      -
        key: "EPH (m)"
        value: "0.94"
      -
        key: "EPV (m)"
        value: "1.46"
  -
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "udp://:14650@"
    values:
      -
        key: "Heartbeats since startup"
        value: "1867"
      -
        key: "Frequency (Hz)"
        value: "0.963034"
      -
        key: "Vehicle type"
        value: "Fixed wing aircraft"
      -
        key: "Autopilot type"
        value: "ArduPilot - Plane/Copter/Rover/Sub/Tracker"
      -
        key: "Mode"
        value: "RTL"
      -
        key: "System status"
        value: "Critical"
  -
    level: 2
    name: "mavros: System"
    message: "Sensor helth"
    hardware_id: "udp://:14650@"
    values:
      -
        key: "Sensor present"
        value: "0x0361FC3F"
      -
        key: "Sensor enabled"
        value: "0x0061FC3F"
      -
        key: "Sensor helth"
        value: "0x0171FC2F"
      -
        key: "3D gyro"
        value: "Ok"
      -
        key: "3D accelerometer"
        value: "Ok"
      -
        key: "3D magnetometer"
        value: "Ok"
      -
        key: "absolute pressure"
        value: "Ok"
      -
        key: "differential pressure"
        value: "Fail"
      -
        key: "GPS"
        value: "Ok"
      -
        key: "3D angular rate control"
        value: "Ok"
      -
        key: "attitude stabilization"
        value: "Ok"
      -
        key: "yaw position"
        value: "Ok"
      -
        key: "z/altitude control"
        value: "Ok"
      -
        key: "x/y position control"
        value: "Ok"
      -
        key: "motor outputs / control"
        value: "Ok"
      -
        key: "rc receiver"
        value: "Ok"
      -
        key: "AHRS subsystem health"
        value: "Ok"
      -
        key: "Terrain subsystem health"
        value: "Ok"
      -
        key: "CPU Load (%)"
        value: "2.4"
      -
        key: "Drop rate (%)"
        value: "0.0"
      -
        key: "Errors comm"
        value: "0"
      -
        key: "Errors count #1"
        value: "0"
      -
        key: "Errors count #2"
        value: "0"
      -
        key: "Errors count #3"
        value: "0"
      -
        key: "Errors count #4"
        value: "0"
  -
    level: 1
    name: "mavros: Battery"
    message: "Low voltage"
    hardware_id: "udp://:14650@"
    values:
      -
        key: "Voltage"
        value: "0.00"
      -
        key: "Current"
        value: "-0.0"
      -
        key: "Remaining"
        value: "-1.0"
  -
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "udp://:14650@"
    values:
      -
        key: "Timesyncs since startup"
        value: "18554"
      -
        key: "Frequency (Hz)"
        value: "9.926702"
      -
        key: "Last RTT (ms)"
        value: "24.926647"
      -
        key: "Mean RTT (ms)"
        value: "28.996867"
      -
        key: "Last remote time (s)"
        value: "1890.591225000"
      -
        key: "Estimated time offset (s)"
        value: "0.000000000"

```

### Check ID

```
rosrun mavros checkid

OK. I got messages from 1:1.

---
Received 1112 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 22, 193, 136, 11020, 152, 150, 24, 27, 29, 30, 32, 33, 163, 36, 165, 168, 42, 178, 182, 62, 65, 74, 87, 111, 241, 242, 116, 125

Jaeyoung-Lim, 2019-03-05:
You need to send setpoints within the timeout to enter offboard mode. Please refer to [this](http://dev.px4.io/en/ros/mavros_offboard.html) document.

vooon, 2019-03-05:
ArduPilot do not have OFFBOARD mode, it has GUIDED.

alexisutc, 2019-03-05:
thx for the answer.

Eventually i only want to control manually my PWM's, so i thougth that using actuator_control could solve my problem. Can guided-mode allow me to use the actuator_control node ?

vooon, 2019-03-05:
Initially it was PX4-only feature, but i don't know current state of APM.

vooon, 2019-03-05:
If you only want to control some servo channel, you can use command plugin to send SET_SERVO cmds.

alexisutc, 2019-03-05:
> 
> 
> If you only want to control some servo channel, you can use command plugin to send SET_SERVO cmds.

how ?

Jaeyoung-Lim, 2019-03-05:
@alexisutc It is written on your log
```
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "udp://:14650@"
    values:
      -
        key: "Heartbeats since startup"
        value: "1867"
      -
        key: "Frequency (Hz)"
        value: "0.963034"
      -
        key: "Vehicle type"
        value: "Fixed wing aircraft"
      -
        key: "Autopilot type"
        value: "ArduPilot - Plane/Copter/Rover/Sub/Tracker"
      -
        key: "Mode"
        value: "RTL"
      -
        key: "System status"
        value: "Critical"
```

alexisutc, 2019-03-06:
Ok thanks

since I use the APM with arduplane i will forget about the offboard mode and focus on the plugin you told me about.

alexisutc, 2019-03-06:
Hi again,

after some researches i figured folowing out :
you say to use SET_SERVO via mavlink.
I can't figured how to write on the node. I guess it should looks like :

rostopic pub /mavlink/to mavros_msgs/Mavlink "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
framing_status: 0
magic: 0
len: 2
incompat_flags: 0
compat_flags: 0
seq: 0
sysid: 1
compid: 250
msgid: 183
checksum: 0
payload64: [0]
signature: ''"

On the other hand I can write on the overrideRCIN working just fine but with a latency of 2 seconds ! Any idear why ?

vooon, 2019-03-06:
I said to use command plugin, not trying to interact with /mavlink topics.
```
rosrun mavcmd long 183 1 1500 0 0 0 0 0
```
https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_SERVO

alexisutc, 2019-03-06:
hello voon

thx but mavcmd doesnt exist, how to install ?
`[rospack] Error: package 'mavcmd' not found`
Also I just would like to have a node on which i publish a pwm and that applies it on the corresponding PWM bus.

amalpavithran, 2019-06-14:
Its actually
`rosrun mavros mavcmd`