mavlink/mavros #1182: Low update rate on Mavros topic

lucaspenna00, 2019-02-19:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
Hello everyone,
I'm working on a indoor drone project and recently I noticed something bad: 
the topics are being updating in a very low rate (less than 1Hz) (including the topic mavros/local_position/pose that is important to control the drone on OFFBOARD mode).
It's turning the flight control difficult by setposition topic because the drone doesn't know its correct altitude in real time. When I test this test algorithm (fly 1 meter and hold its position), on Gazebo Simulator, the Mavros topics are updating in a nice rate.
In order to verify if the problem is the telemetry system, I tested the information update rate on QGroundControl and when I move the drone, the information are instantaneously updated. Then, I suppose the telemetry system (3DR Radio Link) is not the problem.

I'm launching mavros by this command:

roslaunch mavros px4.launch fcu_url:=/dev/ttyUSB0:57600
 
If there's a way to solve this problem (high latency on topic updating using telemetry) please tell me.

### MAVROS version and platform

Mavros: 0.18.4
ROS: Lunar 1.13.7
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 3.7.1 (PX4FMU_V3) Pixhawk 2

### Node logs

started roslaunch server http://lucas-Inspiron-7460:40689/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
[... 338 lines cut ...]
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last dt (ms)"
        value: "9.376617"
      - 
        key: "Mean dt (ms)"
        value: "0.000000"
      - 
        key: "Last system time (s)"
        value: "3023.959504000"
      - 
        key: "Time offset (s)"
        value: "1550615055.404127836"

### Check ID

OK. I got messages from 1:1.

Received 371 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 36, 230, 32, 105, 74, 141, 77, 111, 241, 147, 245, 4, 24, 132, 106, 30

Thanks for the help!

TSC21, 2019-02-20:
With that baud rate in that telemetry link it looks perfectly normal that you are getting the data at those rates, considering that the link rate profiles of PX4 for these kind of links are quite low. I would follow https://docs.px4.io/en/peripherals/serial_configuration.html and try to setup your link baud rate to 115200.

lucaspenna00, 2019-02-20:
Dear TSC21,

Thank you for reply me.

I followed what you recommended me but I hadn't success. When I change the link baudrate to 115200, I can't access the drone via telemetry. The QGroundControl only displays: "Waiting for Vehicle Connection". I tried to follow that link you posted here and I changed these parameters (aiming to change the baudrate link):

SER_TEL1_BAUD - from "57600" to "115200"
MAV_0_RATE - from "1200" to "115200"
MAV_0_MODE - from "0" to "1" (normal -> custom)

If there's something wrong in this procedure, please, tell me. 

Thank you very much.

TSC21, 2019-02-20:
Please bring this matter to PX4 Slack.

closed 2019-02-20 by TSC21

lucaspenna00, 2019-03-05:
I solved this issue just changing the parameter "MAV_0_RATE" (from 1200 (default) to 57600) without change baud rate parameters.

afonsoVale, 2020-10-28:
> I solved this issue just changing the parameter "MAV_0_RATE" (from 1200 (default) to 57600) without change baud rate parameters. Previously I was trying to change the SER_TEL1_BAUD parameter (from 57600 to 115200) but it didn't solve the issue and the telemetry didn't work with this baud rate. Then I tested just the MAV_0_RATE and it worked well.

Hi, how did you go about changing the MAV_0_RATE? I've looked for it in the parameter list in Mission Planner but can't find it.

lucaspenna00, 2020-12-07:
> Hi, how did you go about changing the MAV_0_RATE? I've looked for it in the parameter list in Mission Planner but can't find it.

It depends on the PX4 version you are running. I accessed this parameter by QGroundControl. You can check the parameters here:

https://docs.px4.io/v1.9.0/en/advanced_config/parameter_reference.html

If you are not finding the parameter by QGroundControl I recommend you set this parameter by PX4 command line

aihamalharbat, 2020-12-10:
For anyone who is still looking for a solution.

I wanted to increase the publishing rate of Odometry & HIGHRES_IMU messages.

Here's how I solved this problem:

1- Increase the baud rate of the serial communication by changing these PX4 parameters on QGroundControl, and then reboot the UAV. 
```
SER_TEL1_BAUD - from "57600" to "115200" 
MAV_0_RATE - from "1200" to "115200" 
MAV_0_MODE - from "0" to "1" (normal -> custom) 
```
References:

https://docs.px4.io/master/en/peripherals/serial_configuration.html#how-to-configure-a-port 

https://github.com/mavlink/mavros/issues/1182#issuecomment-465672331 

2- Change the rate that Mavlink publishes the ODOMETRY/ HIGHRES_IMU messages in. This can be done by writing this command on the microSD card in PX4. How? 

- Remove the card from Pixhawk and connect to your PC. Then go to ./fs/etc/extras.txt 

- If this directory (or part of it) does not exist, create it. Then, inside extras.txt, we should write: 
```
mavlink stream -d /dev/ttyS0 -s HIGHRES_IMU -r 200 
mavlink stream -d /dev/ttyS0 -s ODOMETRY -r 200 
```
Reference:
https://dev.px4.io/v1.9.0/en/concept/system_startup.html#starting-additional-applications 

For completeness, here is my setup details:
#### MAVROS version and platform 

Mavros: 1.4 

ROS: Melodic 

Ubuntu: 18.04 

#### Autopilot type and version  

PX4 

Version: v1.10.1 

PX4_FMU_V5 (V500) 

NuttX, v7.29.0

mengchaoheng, 2023-04-22:
@ayhamalharbat @lucaspenna00 @TSC21 can you help me please! https://github.com/PX4/PX4-Autopilot/issues/21504

Mu99-M, 2023-05-19:
@ayhamalharbat I tried this, but it didn't work for me.
Has anyone found a solution?