mavlink/mavros #1185: Excessive CPU load (~100% over 8 threads) for APM + MAVROS inside Docker container

ChrisTimperley, 2019-02-27:
### Issue details

I'm trying to run connect MAVROS to the ArduPilot's SITL from inside a Docker container (see attached for `Dockerfile`). APM runs without any problems inside the container. However, when I launch MAVROS, I observe a recurring error in the node logs, and the MAVROS consumes all available CPU resources.

[image]

Interestingly, CPU usage also appears to be fairly high when only `roscore` is running.

[image]

### MAVROS version and platform

MAVROS: 0.29.0
ROS: Kinetic / Lunar (occurs on both)
Container OS: Ubuntu 16.04 / 17.04
Host OS: Ubuntu 18.04 / Arch-2019.02.01 (same machine, different OS)
Machine: 14" Razer Blade, 2016.  i7-6700HQ (4c/8t), 16GB DDR4, GTX 1060.

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 3.6.4 (https://github.com/ArduPilot/ardupilot/releases/tag/Copter-3.6.4)

### Node logs

```
... logging to /root/.ros/log/9d20e3c8-3a3e-11e9-907f-0242ac110002/roslaunch-9d820d31bf83-260.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://9d820d31bf83:45085/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
[... 482 lines cut ...]
```

### Bug Reproduction

Using the provided [Dockerfile](https://github.com/mavlink/mavros/files/2908434/Dockerfile.txt):

```
$ docker build -t mavros-timing-bug .
$ docker run --rm -it mavros-timing-bug
~ roscore &
...

~ /ros_ws/src/ArduPilot/build/sitl/bin/arducopter --model copter &
Creating model copter at speed 1.0
Home: -35.363261 149.165235 alt=584.000000m hdg=353.000000
Starting sketch 'ArduCopter'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....

~ roslaunch mavros apm.launch fcu_url:=tcp://127.0.0.1:5760@5760
...
```

ChrisTimperley, 2019-02-27:
Possibly related: https://github.com/ros/ros_comm/issues/1343

vooon, 2019-02-27:
Do you use prebuilt packages or from source?
Because roscore also eats CPU, i think that most likely problem in roscpp.
And yes, bug in node::spin may result in high CPU usage.

Can you try older distribution? Mavros shoud still work on Indigo (i don't recommend to use it, but it help with understanding).

ChrisTimperley, 2019-02-27:
> Do you use prebuilt packages or from source?

Packages are built from source.

> Can you try older distribution? Mavros shoud still work on Indigo (i don't recommend to use it, but it help with understanding).

The bug doesn't occur on Indigo.

hexray-newbee, 2021-08-03:
ROS foxy also occurs.

vooon, 2021-08-03:
Closing that issue as stall. ROS2 is completely different story.

closed 2021-08-03 by vooon