mavlink/mavros #1188: RC control with mavros and sitl gazebo

Pedminacar, 2019-02-28:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi,
Can anyone tell me what is necessary for mavros to publish to rc/out? The current situation is as follows: I launch mavros, px4, sitl gazebo in a launch file then I try to publish inputs through rc/override, but when I echo the rc/out, the values (900) remain constant. I realized that happens to more commands, I mean, I insert an input and nothing happens, no output. I also realized that when I use "commander arm" or "commander takeoff", the rc/out values change but for some reason with rc/override nothing happens. I think there is some sort of parameter that is preventing mavros to calculate and send information to drone. My goal is to use a PS4 controller to guide my drone in gazebo, so I need to somehow send pwm values to drone.

Help is very appreciated.

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Melodic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: ?3.7.1?

### Node logs

```
... logging to /home/pncarvalho/.ros/log/8ae7283c-3bad-11e9-ae6e-5076afbdc004/roslaunch-PDM-00865-3761.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/pncarvalho/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://PDM-00865:34927/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
[... 477 lines cut ...]
      - 
        key: "Frequency (Hz)"
        value: "10.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.038676"
      - 
        key: "Last remote time (s)"
        value: "198.236000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.163956405"

```

### Check ID

```
It gives me error
![screenshot from 2019-02-28 23-14-37](https://user-images.githubusercontent.com/48107157/53605451-72a14d00-3baf-11e9-9166-7a50104d9ae0.png)

```

TSC21, 2019-03-01:
PX4 doesn's support `RC_CHANNELS_OVERRIDE` as input. You need to use `MANUAL_CONTROL`. Check the `manual_control` plugin.

closed 2019-03-01 by TSC21

Pedminacar, 2019-03-01:
Hi,
I tried to publish to manual_control/send but the rc/out/doesnt change its values. Plus I don't receive anything in manual_control/control.

TSC21, 2019-03-04:
@Pedminacar actually I refactor: PX4 does support RC override, but you need to make sure that the target system is the same as the system ID of the vehicle: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1747