mavlink/mavros #1190: diagnostics rate not periodic while logging pixhawk cpu usage

siddhu45, 2019-03-04:
Hi guys,

I am trying to log pixhawk CPU usage using the rostopic `/diagnostics`. But even though it says it publishes at 1Hz, when I subscribe to this topics to extract the CPU usage, most of the messages received by the subscriber in the node does not contain that much info, and I get the error `segmentation fault (core dumped)`. I checked the vector length and see that it receives full message only sporadically. There is not relation in between the timestamps I receive, so I have to use a workaround like:

```
#include <ros/ros.h>
#include <diagnostic_msgs/DiagnosticArray.h>
#include <std_msgs/Float32.h>

ros::Publisher pixhawk_cpu_usage_pub;
ros::Subscriber pixhawk_diagnostics_sub;
std_msgs::Float32 pub_msg;

void diagnostics_cb(const diagnostic_msgs::DiagnosticArray::ConstPtr& msg) {
    if (msg->status.size() >= 4) {
        // std::cout << msg->status[3].values[3].key.c_str() << std::endl;
        pub_msg.data = atof(msg->status[3].values[3].value.c_str());
        pixhawk_cpu_usage_pub.publish(pub_msg);
    }
}

int main(int argc, char** argv){
    ros::init(argc, argv, "pixhawkCPUpub");
    ros::NodeHandle nh;
    pixhawk_diagnostics_sub = nh.subscribe("/diagnostics", 1, diagnostics_cb);
    pixhawk_cpu_usage_pub = nh.advertise<std_msgs::Float32>("/pixhawk_cpu_usage", 10);
    ros::spin();
}
```

But when I use `rostopic echo /diagnostics` I can see all the content. Can someone explain why part of the message is not published regularly or is there something I am missing. thanks. It is not something critical but as I am logging I just want to know if there is a consistent way to log this parameter.

Thanks

### MAVROS version and platform

Mavros: ?0.29.0?
ROS: ?Kinetic?
Ubuntu: ?16.04?

vooon, 2019-03-04:
`/diagnostics` are used by all nodes that wants to emit diag data. So some nodes may publish different amount of diag entries. That is similar to how `/tf` works.
You should not rely on indexes, they may change. You should walk and compare names.

siddhu45, 2019-03-05:
ok, got it. thanks
So there is actually no specific time time on which they will publish data. I will do as you suggested
Thanks