mavlink/mavros #1193: FCU not connected Heartbeat not received

lilnpuma, 2019-03-05:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
I'm able to view all the published topics using _rostopic list_  command 
but we are not able to get data as we use _rostopic echo_ command 
we see usb is connected by using _lsusb_ command and found out that the mount location was /dev/ttyACM0

### MAVROS version and platform

Mavros: ?0.27.0?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[ - ] ArduPilot
[  ] PX4

Version: ?3.6.5?

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyACM0:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
[... 1138 lines cut ...]
  File "/usr/lib/python2.7/httplib.py", line 897, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 859, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 836, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
error: [Errno 111] Connection refused
```
State Log
```
header: 
  seq: 49
  stamp: 
    secs: 1551766836
    nsecs: 345778880
  frame_id: ''
connected: False
armed: False
guided: False
mode: ''
system_status: 0
---
```

vooon, 2019-03-05:
Do you connect via USB port on FCU? Usually only that uses CDC ACM protocol, which maps to ttyACM%d.
All USB-Serial chips should be on ttyUSB%d.

For ACM baud rate is ignored, for serial ports - default is 57600.
Check `dmesg`. Also do you sure that firmware is flashed and working ok? Try to connect with GCS.

YCH188, 2019-03-08:
I have also encountered the problem, at that time, the problem is the FCU is broken.

maxammann, 2019-04-24:
Having the same problem, currently trying to debug it.

YCH188, 2019-04-25:
@maxammann I have solved it. Use make px4_sitl_default gazebo instead of make posix_sitl_default gazebo. It seems that the px4 team changed firmware.

maxammann, 2019-04-25:
@YCH188 so you are having problems with the simulator?
I'm having this problem with a Pixracer PX4, so with real hardware. I'm experiencing the above problem about 50% of the time when launching mavros.

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248

MaximeAubanel, 2020-07-01:
Any news on this issue ? @maxammann Have you find a solution ? I'm having the same issue

maxammann, 2020-07-03:
If I remember correctly, then we were having trouble with the serial connection between the PX4 and a Raspberry PI.

We forgot to disable bluetooth (disable-bt) and disable the hci service (sudo systemctl disable hciuart): https://www.raspberrypi.org/documentation/configuration/uart.md

Double check if you are using a PI :P

MaximeAubanel, 2020-07-04:
On my side, it was that MAV_SYS_ID of the autopilot wasn't matching tgt_system of mavros.

I had to run mavros that way:

```
roslaunch mavros px4.launch fcu_url:="$FCU_URL" tgt_system:="$MAV_SYS_ID"
```

YifeiNie, 2024-10-30:
> On my side, it was that MAV_SYS_ID of the autopilot wasn't matching tgt_system of mavros.
> 
> I had to run mavros that way:
> 
> ```
> roslaunch mavros px4.launch fcu_url:="$FCU_URL" tgt_system:="$MAV_SYS_ID"
> ```

It work for me, tks!

invidia0, 2025-03-10:
> On my side, it was that MAV_SYS_ID of the autopilot wasn't matching tgt_system of mavros.
> 
> I had to run mavros that way:
> 
> ```
> roslaunch mavros px4.launch fcu_url:="$FCU_URL" tgt_system:="$MAV_SYS_ID"
> ```

This solved my issue as well.