mavlink/mavros #1201: libmavconn: recursion in udp.cpp crashing system when no interface is up

ArkadiuszNiemiec, 2019-03-07:
### Issue details

I have removed a NetworkManager form my Raspbian based RPi and run a mavros with GCS URL defined. The result is that I have a flood of warnings generated by https://github.com/mavlink/mavros/blob/a7ef4fc0ec153307cbce3c98a998956c6296b954/libmavconn/src/udp.cpp#L279 which is not that bad but the problem is that after few seconds my system just crashed or kernel paniked. I have tracked it to the recursive calling of `void MAVConnUDP::do_sendto(bool check_tx_state)` by this line: https://github.com/mavlink/mavros/blob/a7ef4fc0ec153307cbce3c98a998956c6296b954/libmavconn/src/udp.cpp#L302

Is this infinite recursion intended? Is there a know workaround to have a GCS URL defined with unstable interface?

### MAVROS version and platform

Mavros: 0.29.0
ROS: ?Melodic
Rasbian Stretch

### Autopilot type and version

PX4 1.9

vooon, 2019-03-07:
That is intentional: https://github.com/mavlink/mavros/commit/f1b34d727940e2dd7c2be99b72e43961292a8d1b#diff-02eba5461689ed95e6dc463ccca270fb

But should not lead to kernel panics. Perhaps something wrong in environment.

ArkadiuszNiemiec, 2019-03-07:
We have only few MB of RAM avaliable and swapping on SD Card is probably killing it. I will test it tomorrow.

Do you know if I can just drop the message instead of retrying? I think that its better to do so instead of trying to sending it indefinitely.

Is it blocking the GCS plugin or is it trying to send more and more messages? I have to investigate it further...

rk800506, 2021-02-19:
Is there a solution for the error?

rk800506, 2021-02-21:
@vooon, @ArkadiuszNiemiec , I also faced the same issue while i ran PX4.launch in mavros package.
What I encountered is that I just replaced gcs_url with only "" (nothing within the quotes as I am not running any GCS like qgc or anything else). After this operation, error stopped appearing.

But, when I tried to reproduce the error by giving gcs_url = "udp://:14550@xxx.xx.xx.xx.:14457", the error did not come. SO, could not know the source of error but since I am not running any GCS so this trick solved the problem.