mavlink/mavros #1203: Received sensor data has different orientation

Seanmatthews, 2019-03-08:
### Issue details

It's unclear how to define the sensor orientation as it's coming from Gazebo, such that this error does not occur:

> [/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]

You can reproduce it by running `mavros_posix_sitl.launch` with the `typhoon_h480` vehicle model (defined in sitl_gazebo) that uses the sonar sensor.

### MAVROS version and platform

Mavros: Kinetic apt repo latest
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x] PX4

Version: 1.8.2

### Node logs
A sample:

> [/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
> [/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
> [/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
> [Wrn] [msgs.cc:1852] Conversion of sensor type[gpu_ray] not supported.
> [Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
> [Wrn] [msgs.cc:1852] Conversion of sensor type[gpu_ray] not supported.
> [Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
> [/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
> [/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
> [/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
> [/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]

### Diagnostics
```
header: 
  seq: 25
  stamp: 
    secs: 4239
    nsecs: 652000000
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Received packets:"
        value: "7368"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "199"
      - 
        key: "Tx sequence number:"
        value: "155"
      - 
        key: "Rx total bytes:"
        value: "297097"
      - 
        key: "Tx total bytes:"
        value: "10196"
      - 
        key: "Rx speed:"
        value: "17634.000000"
      - 
        key: "Tx speed:"
        value: "547.000000"
  - 
    level: 0
    name: "mavros: GPS"
    message: "3D fix"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Satellites visible"
        value: "10"
      - 
        key: "Fix type"
        value: "3"
      - 
        key: "EPH (m)"
        value: "0.00"
      - 
        key: "EPV (m)"
        value: "0.00"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "17"
      - 
        key: "Frequency (Hz)"
        value: "1.000000"
      - 
        key: "Vehicle type"
        value: "Quadrotor"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "AUTO.LOITER"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000020"
      - 
        key: "Sensor enabled"
        value: "0x00000000"
      - 
        key: "Sensor helth"
        value: "0x00200020"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Voltage"
        value: "12.15"
      - 
        key: "Current"
        value: "-1.0"
      - 
        key: "Remaining"
        value: "100.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "177"
      - 
        key: "Frequency (Hz)"
        value: "10.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.225988"
      - 
        key: "Last remote time (s)"
        value: "17.521307000"
      - 
        key: "Estimated time offset (s)"
        value: "4222.030393887"
```
---

### Check ID

```
OK. I got messages from 1:1.

---
Received 6499 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 132, 140, 141, 147, 22, 24, 4, 26, 30, 31, 32, 33, 44, 74, 76, 77, 83, 85, 87, 230, 105, 111, 241, 242, 245```

TSC21, 2019-03-08:
The config of the Mavlink DISTANCE_SENSOR msg has to be done on the sitl_gazebo side.

closed 2019-03-08 by TSC21

Seanmatthews, 2019-03-08:
@TSC21 Can you expand? As far as I can tell, the orientation is configured on the sitl_gazebo side and it's not passing through. I added to the typhoon_h480 SDF model a pose for the sonar sensor:

>     <include>                                                                                                                                                                                                                                                         
>       <uri>model://sonar</uri>                                                                                                                                                                                                                                        
>       <pose>0.12 0 0 0 4.71239 0</pose>                                                                                                                                                                                                                               
>     </include>

TSC21, 2019-03-08:
Check the `gazebo_mavlink_interface` of sitl_gazebo. This is not a MAVROS related topics so should be discussed on the appropriate forums.

AlexWUrobot, 2022-12-20:
After commented “distance sensors” and “range finders” on px4_pluginlists.yaml, I finally can get the distance_sensor rostopic 
ref: https://discuss.px4.io/t/how-to-read-distance-sensor-data-from-mavros-topic/21944 
However, I faced the similar situation, when running
roslaunch px4 mavros_posix_sitl.launch
 
There is the error:
DS: hrlv_ez4_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!
 

[image]