mavlink/mavros #1208: mavlink::common::msg::ODOMETRY is not sending out for somereason

limhyon, 2019-03-28:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

We are testing visual odometery. Our SLAM node export nav_msgs/Odometry, so we remap the IO to /mavros/odometry/out.

We can echo the message, however, it won't be available in the FMU side.
We've checked it by printing message id in 

https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L133

### MAVROS version and platform

Mavros: today's master.
ROS: Kinetic
Ubuntu: 16.04, Jetson Ubuntu.

### Autopilot type and version

[  ] ArduPilot
[ V ] PX4

Version: 1.9.0 (January version)

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

I will do if further action is required.

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

I will do if further action is required.

```
rosrun mavros checkid
```

vooon, 2019-03-28:
I don't see any possibility to drop ODOMETRY in plugin.
But first: please check connectivity by RQT (Graph plugin), try to subscribe.
Then also try to enable debug log for odom plugin (it also possible to do in runtime via RQT).

As i remember not while ago ODOMETRY definition was changed.
If you have different message on FCU it will be dropped as corrupted (silently!).

CC: @TSC21

limhyon, 2019-03-28:
I've checked FCU, and it has odometry mavlink topic ODOMETRY. But, handler is not called. Still figuring out why.

TSC21, 2019-03-28:
@limhyon:

>As i remember not while ago ODOMETRY definition was changed.
If you have different message on FCU it will be dropped as corrupted (silently!).

It means that even if you have the handler there, if the definition is wrong, is not going to work. What PX4 version are you using? If you are using upstream master you are going to notice that the velocity covariance field has changed from `twist_covariance` to `velocity_covariance`, which matches the [`ODOMETRY` message definition](https://mavlink.io/en/messages/common.html#ODOMETRY).

TSC21, 2019-05-31:
@limhyon updates?