mavlink/mavros #1209: On takeoff: Request failed. Check mavros logs. ACK: 4

Hani-H, 2019-04-02:
### Issue details

I'm trying to fly my quadrotor indoors with mavros. The drone arms normally, and it did manage to take off before although I'm not sure what changed this time.

Running `rosrun mavros mavcmd takeoff 0 0 0 0 1`:
> Request failed. Check mavros logs. ACK: 4

Running `rosrun mavros mavcmd takeoffcur 0 0 1`:
> Something went wrong. Topic timed out.

I'm following the drone manufacturer's official instructions:
https://edu.gaitech.hk/gapter/gapter-ros-tutorials.html

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Indigo
Ubuntu: 14.04

### Autopilot type and version

[ - ] ArduPilot
[  ] PX4

Version: 3.4.6

### Node logs

```
gapter@gapter:~$ roslaunch ros_gapter bringup_minimal.launch 
... logging to /home/gapter/.ros/log/e79155c6-53a6-11e2-a7ed-0013ef400f1a/roslaunch-gapter-1880.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.1:42532/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/fcu_url: /dev/ttyUSB0:57600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['actuator_contro...
[... 233 lines cut ...]
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 30 messages, from 2 addresses
sys:comp   list of messages
 255:0     0
  1:1     0
```

amalpavithran, 2019-06-14:
I was also facing a similar issue. I solved the issue my keeping my RC on and giving some throttle after giving the takeoff command. It still gives me the error but the motors go full throttle. Haven't tested with props.

Nikhiljohnson21, 2023-08-31:
did you find a solution for this error, i am also facing the same error