mavlink/mavros #1210: pyyaml fails to parse YAML files

julianoes, 2019-04-03:
On PX4 SITL launch using roslaunch, pyyaml fails to parse the px4_config.yaml file.

### MAVROS version and platform

Mavros: 0.29.2
ROS: Melodic
Ubuntu: 18.04.2

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 99fb88ce839879e35f86c199f7dc50fa361f85e4

### Diagnostics

```
roslaunch px4 multi_uav_mavros_sitl.launch
... logging to /home/julianoes/.ros/log/20d6678a-55f9-11e9-a624-080027ca140c/roslaunch-ubuntu1804-vbox-1823.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/loader.py:412: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
RLException: error loading <rosparam> tag: 

could not determine a constructor for the tag '!degrees'                                                                
  in "<string>", line 63, column 27:                                                                                    
      angular_velocity_stdev: !degrees 0.02                                                                             
                              ^                                                                                         
XML is <rosparam command="load" file="$(arg config_yaml)"/>                                                             
The traceback for the exception was written to the log file

```

The same issue is also captured here:
https://github.com/PX4/Firmware/issues/11662

vooon, 2019-04-04:
Looks like something changed into yaml library.
Previously roslaunch provide `degrees` tag, that simply converts value to radians.
As a temporary fix you may replace it with `0.00034906585039886593`.

julianoes, 2019-04-04:
Ok thanks. What do you suggest as the real long term fix?

vooon, 2019-04-04:
Well, it looks like that roslaunch should be updated to handle pyyaml changes.

Pedro-Roque, 2019-07-05:
Also got to this error. @julianoes what did you end up doing?

julianoes, 2019-07-08:
I did this and it works but I don't think it's correct: https://github.com/PX4/Firmware/issues/11662#issuecomment-479470281

Pedro-Roque, 2019-07-08:
> I did this and it works but I don't think it's correct: [PX4/Firmware#11662 (comment)](https://github.com/PX4/Firmware/issues/11662#issuecomment-479470281)

I ended up converting from degrees to radians ahah ^^ I remember someone saying on the dev summit that we should be using rads all the time, so just went with that approach

julianoes, 2024-03-04:
Too old, closing.

closed 2024-03-04 by julianoes