mavlink/mavros #1211: GCS heartbeat instead of Onboad  heartbeat

hardeman, 2019-04-03:
Hi there,

Is it possible to add the option to start px4.launch with a GCS heartbeat? Otherwise the PX4 software won't send message #253 (statustext) over mavlink and the topic /mavros/statustext/recv stays empty.

Thank you in advance.

Regards,
Erik

vooon, 2019-04-03:
Yes, it is possible: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/sys_status.cpp#L453

hardeman, 2019-04-04:
Oke, thank you. But I don't see how I can use that to launch mavros with the right arguments. The only arguments I can find are these:

        <arg name="fcu_url" default="/dev/ttyACM0:57600" />
	<arg name="gcs_url" default="" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />
	<arg name="fcu_protocol" default="v2.0" />
	<arg name="respawn_mavros" default="false" />

vooon, 2019-04-04:
Via config: https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml

hardeman, 2019-04-04:
I added heartbeat_mav_type to px4_config.yaml like this:

conn:
  heartbeat_mav_type: "MAV_TYPE_GCS"  
  heartbeat_rate: 1.0    # send hertbeat rate in Hertz
  timeout: 10.0          # hertbeat timeout in seconds
  timesync_rate: 10.0    # TIMESYNC rate in Hertz (feature disabled if 0.0)
  system_time_rate: 1.0  # send system time to FCU rate in Hertz (disabled if 0.0)

I also tried: "mav_type_gcs", mav_type_gcs, 6, "6"  but nothing seems to work. Is there something I'm doing wrong?

hardeman, 2019-04-04:
The only way I don't get an error at startup of mavros, is when I use 6.

But the original problem still exists because /mavros/statustext/recv is still empty.

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248

hardeman, 2020-02-26:
This was an issue with older versions of PX4, but in the latest release it's not an issue anymore.

closed 2020-02-26 by hardeman

shupx, 2024-09-17:
Try to set the mavros `conn/heartbeat_mav_type`:
```
	<arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
	<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />
	<node pkg="mavros" type="mavros_node" name="mavros" required="false" clear_params="true" output="screen" respawn="false">
		<param name="fcu_url" value="$(arg fcu_url)" />
		<param name="gcs_url" value="" />
		<param name="target_system_id" value="$(arg ID)" />
		<param name="target_component_id" value="1" />
		<param name="fcu_protocol" value="v2.0" />

		<!-- load blacklist, config -->
		<rosparam command="load" file="$(arg pluginlists_yaml)" />
		<rosparam command="load" file="$(arg config_yaml)" />
		
		<param name="conn/heartbeat_mav_type" value="GCS"/> <!--added  to receive statustext-->
	</node>
```