mavlink/mavros #1212: MAVROS GPS RTK

KonstantineMushegian-TRI, 2019-04-05:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi,

I'm working on integrating the Emlid GPS RTK to be used with mavros on a Rover. I've set up all the hardware and am able to get the rtk working through Mission Planner.

However, when I try to use mavros gps_rtk plugin, I am not getting the GPS status to be rtk. I know ROS doesn't support GPS statuses, so I've changed mavros a bit to pass up a status from raw_fix in global_position.cpp.

Here's how I am listening to RTCP corrections and publishing on the `send_rtcm` topic:

```
def listen_tcp_and_publish_fix():
	s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
	TCP_IP = socket.gethostbyname(TCP_HOSTNAME)
	s.connect((TCP_IP, TCP_PORT))

	while 1:
		data = s.recv(BUFFER_SIZE)
		data = bytearray(data)
		h = Header()
		h.stamp = rospy.Time.now()

		r = RTCM()
		r.header = h
		r.data = list(data)
		rtk_pub.publish(r)
	s.close()
if __name__ == '__main__':
	rospy.init_node('rtk_listener')
	rtk_pub = rospy.Publisher('/sally/mavros/gps_rtk/send_rtcm', RTCM, queue_size=1)
	listen_tcp_and_publish_fix()
```

I'm not even sure where to look next, so any pointers would be appreciated.

### MAVROS version and platform

Mavros: 0.29.2
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: Latest

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

KonstantineMushegian-TRI, 2019-04-05:
The console output looks normal, I'm using a GPS that came with our Pixhawk as GPS 1, and the Emlid Reach M+ as GPS2.

Since I was able to get RTK injection through Mission Planner I'm guessing my hardware setup & autopilot parameters are correct.

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248