mavlink/mavros #1214: /mavros/setpoint_raw/local frame is not match with yaw of mavros/imu/data

mandu46, 2019-04-09:
----

### Issue details

I am testing /mavros/setpoint_raw/local in gazebo and in real world.
I want to control velocity of UAV by giving velocity value at /mavros/setpoint_raw/local/velocity
But when I give velocity.x value, UAV moves to right, and moves forward when I give  velocity.y. 
I checked frame of /mavros/local_position/odom is ENU and orientation of /mavros/imu/data also has a ENU frame. I think x axis is forward in body frame, but /mavros/setpoint_raw/local shows differently. It have (x:right, y: forward, z: up) frame. The coordinate_frame value in /mavros/setpoint_raw/local is 8 (FRAME_BODY_NED), is it relevant? 
I want to set body frame which have forward x axis, up z axis with correct yaw value. 
Please help me solving of this problem. 

### MAVROS version and platform

Mavros: 0.18.0
ROS: Kinetic?
Ubuntu: 16.04?

### Autopilot type and version

[  ] ArduPilot
[ V ] PX4

Version: ?3.7.1?

vooon, 2019-04-10:
As i remember, IMU should be in body frame, while local wants ENU (with declaration of NED frame in field, confusion is because mavros does transform ENU-NED, but enum have same names as in FCU).

mandu46, 2019-04-11:
Thanks for reply @vooon . What I confused is default definition of body frame. If body frame is configured as forward for x axis, yaw will be 0  degree when I put UAV`s direction to east and it is equal to when I test with real environment. But at this situation UAV moves to right when I give x axis velocity command and moves to forward when I give y axis velocity command. That makes me confusion of body frame definition.

sergioma295, 2019-06-07:
I am having a big problem. I have created a new issue. Can somebody help me please? #1248